linux-nvme/nvme-cli #645: how to pass a file in which the write command will write

babouFomb, 2020-01-17:
Hi,

I'm new on using nvme-cli. I found this example command `echo "hello world" | nvme write /dev/nvme0n1  --data-size=520 --block-count=0` in the issue #223. Where is stored the data that have been write?

I have read the options of `write` , but I don't know how I can specify the output file (a binary file for example on my SSD) on which the `write` command must write.

Thanks.

keithbusch, 2020-01-17:
You're writing to a raw block interface, not a file interface. The "file" you're writing to is /dev/nvme0n1, which is a special block device.

If you want to write files, don't use this utility, as it is working at a much lower level in the protocol stack below the filesystem abstractions

keithbusch, 2020-01-17:
Another thing, don't use direct nvme writes on any nvme device with formatted data unless you really know what you're doing. This command goes directly to the media, skipping all layers that may understand existing formats, so you may accidentally corrupt that data if you're not careful.

closed 2022-01-19 by igaw