linux-nvme/nvme-cli #648: how to get status code type(SCT) from exit code

ryderhsu0129, 2020-01-21:
I sent a IO command and got the following message:
`NVMe IO command error:unknown(4181)`

and I echo $? to got the status code(SC):
`129`

How do I get status code type(SCT) from the value 4181?thanks!

keithbusch, 2020-01-21:
This format matches nvme cqe status field with the pase shifted out and printed in hex. In 0x4181, 4h is the "do not retry" flag, 1h is the status code type (what you're asking), and 81h (129 in decimal) is the status code.

ryderhsu0129, 2020-01-21:
> This format matches nvme cqe status field with the pase shifted out and printed in hex. In 0x4181, 4h is the "do not retry" flag, 1h is the status code type (what you're asking), and 81h (129 in decimal) is the status code.

@keithbusch 
I get it.
thanks for your reply!

closed 2020-01-21 by ryderhsu0129