linux-nvme/nvme-cli #650: invalid lbaf:255

Neurognostic, 2020-01-21:
The [`nvme-format(1)`](https://github.com/linux-nvme/nvme-cli/blob/master/Documentation/nvme-format.txt) man page reads:

> LBA Format: This field specifies the LBA format to apply to the NVM
>	media. This corresponds to the LBA formats indicated in the
>	Identify Namespace command. Conflicts with --block-size argument.
>	Defaults to 0.

But when I execute:

```console
# nvme format /dev/nvme0
invalid lbaf:255
# echo $?
70
```
or
```console
nvme format /dev/nvme0n1
invalid lbaf:255
# echo $?
70
```

I get the error: `invalid lbaf:255`.

But when I add the option `--lbaf=0` the operation completes successfully.

```console
# nvme format /dev/nvme0 --lbaf=0
You are about to format nvme0, namespace 0xffffffff(ALL namespaces).
Controller nvme0 has child namespace(s):nvme0n1

WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force|-f] option to suppress this warning.
Sending format operation ...
Success formatting namespace:ffffffff
```

#### If `lbaf=0` is the default why do I have to add it explicitly for a successful execution?

```console
# nvme --version
nvme version 1.10.1
```

keithbusch, 2020-01-21:
The man page is outdated. No parameter is supy to use the existing lbaf, but since you sent this command to a controller instead of a namespace, we don't know the existing lbaf.

Neurognostic, 2020-01-21:
Is sending the command to `/dev/nvme0n1` not the equivalent of sending it to the namespace?

keithbusch, 2020-01-21:
Ah, I was looking at your other example /dev/nvme0.

Yes, /dev/nvme0n1 will attempt to send to a namespace's existing format unless the controller does not support this command per-namespace. The tool overrides to the controller level in that case, which requires the lbaf parameter again.

The documentation is pretty old here...

Neurognostic, 2020-01-23:
So when the controller does not support `format` to single namespace(s) specifying `--lbaf` is required?

In my case the fna bit at index `[1:1]` is `0`.

```console
# nvme id-ctrl -H /dev/nvme0n1
...
fna       : 0x5
  [2:2] : 0x1   Crypto Erase Supported as part of Secure Erase
  [1:1] : 0     Crypto Erase Applies to Single Namespace(s)
  [0:0] : 0x1   Format Applies to All Namespace(s)
...
```
This means `--lbaf` is needed?

---

I do not really understand what `lbaf` is and  how to determine appropriate values, I just choose `0` because `nvme-format(1)` said it was the default. I assumed, because it appeared successful, to be the correct value.

The man page says for `--lbaf` *"This corresponds to the LBA formats indicated in the Identify Namespace command."*

I ran `nvme-id-ns(1)` deducing that it was the "Identify Namespace Command", but I am unclear what exactly is the LBA Formats. Below, is it where it says *"Relative Performance: 0 Best (in use)"? `0` best?

With a `--verbose` `nvme-id-ns(1)` command I do see some sort of data structure just below "LBA Format 0:" called `vs[]:`. Are those the "LBA Formats"?

```console
NVME Identify Namespace 1:
nsze    : 0x74706db0
ncap    : 0x74706db0
nuse    : 0x358418
nsfeat  : 0
  [4:4] : 0     NPWG, NPWA, NPDG, NPDA, and NOWS are Not Supported
  [2:2] : 0     Deallocated or Unwritten Logical Block error Not Supported
  [1:1] : 0     Namespace uses AWUN, AWUPF, and ACWU
  [0:0] : 0     Thin Provisioning Not Supported

nlbaf   : 0
flbas   : 0
  [4:4] : 0     Metadata Transferred in Separate Contiguous Buffer
  [3:0] : 0     Current LBA Format Selected

mc      : 0
  [1:1] : 0     Metadata Pointer Not Supported
  [0:0] : 0     Metadata as Part of Extended Data LBA Not Supported

dpc     : 0
  [4:4] : 0     Protection Information Transferred as Last 8 Bytes of Metadata Not Supported
  [3:3] : 0     Protection Information Transferred as First 8 Bytes of Metadata Not Supported
  [2:2] : 0     Protection Information Type 3 Not Supported
  [1:1] : 0     Protection Information Type 2 Not Supported
  [0:0] : 0     Protection Information Type 1 Not Supported

dps     : 0
  [3:3] : 0     Protection Information is Transferred as Last 8 Bytes of Metadata
  [2:0] : 0     Protection Information Disabled

nmic    : 0
  [0:0] : 0     Namespace Multipath Not Capable

rescap  : 0
  [6:6] : 0     Exclusive Access - All Registrants Not Supported
  [5:5] : 0     Write Exclusive - All Registrants Not Supported
  [4:4] : 0     Exclusive Access - Registrants Only Not Supported
  [3:3] : 0     Write Exclusive - Registrants Only Not Supported
  [2:2] : 0     Exclusive Access Not Supported
  [1:1] : 0     Write Exclusive Not Supported
  [0:0] : 0     Persist Through Power Loss Not Supported

fpi     : 0x80
  [7:7] : 0x1   Format Progress Indicator Supported
[... 232 lines cut ...]
0d00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0d10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0d20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0d30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0d40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0d50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0d60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0d70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0d80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0d90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0da0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0db0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0dc0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0dd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0de0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0df0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0e00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0e10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0e20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0e30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0e40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0e50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0e60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0e70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
```

Neurognostic, 2020-01-23:
Sorry for the lengthy question, I really just want to know how I can verify `--lbaf=0` is the appropriate value. Thank you.

keithbusch, 2020-01-23:
Yes, --lbaf=0 is what you'd want to use to format the namespace. It is the only format your device supports.

You can check the "LBA Format" or "lbaf" list at the end of "nvme id-ns ..." to see which formats your drive supports, and use --lbaf=X for the index you want to use. It's a bit of a moot point in your case since it supports just the one format, but it's worth checking since many devices have completely different formats for the same lbaf index.

Neurognostic, 2020-01-24:
Okay, thanks for confirming that `--lbaf=0` is the only option for this particular NVMe.

---

> Yes, /dev/nvme0n1 will attempt to send to a namespace's existing format unless the controller does not support this command per-namespace. The tool overrides to the controller level in that case, which requires the lbaf parameter again.

Could something like this be added to the documentation? An order of operations.

Since there is only one `lbaf` option (`0`) could the tool detect that and use it when pointed at namespace block device (e.g. `/dev/nvme0n1`)?

keithbusch, 2020-01-27:
It's not that there's just the 1 lbaf to choose from. It's that the device says a format to any namespace is a format to all of them, so it's really caution to ensure a user doesn't think they're formatting just one block device and inadvertently wipe them all out. Formats even erase data on unattached namespaces for devices that report their format attributes this way, so it's considered harmful to not require a more explicit command.

You have just the one namespace (as do most pcie attached nvme), so that caution may not necessarily apply here, but it'd be nice if the vendors could also set the appropriate format attributes.

keithbusch, 2020-01-27:
On documentation, I'm very much working on this right now and catching up everything to the 1.4 spec. I'm hoping to wrap it up before usenix vault next month.

closed 2022-01-19 by igaw