linux-nvme/nvme-cli #653: The repository does not have a Realese file.

viktorheli, 2020-01-24:
Good day.
Thank you for the very useful and necessary application.
When updating Ubuntu 18 in Discover, I constantly see this notification:
'https://ppa.launchpad.net/sbates/ppa/ubuntu bionic Release' does not have a release file
Updating from such a repository can't be done securely, and is therefore disabled by default.
See apt-secure(8) manpage fo repository creation and user configuration details.

Can this be fixed? Or is Ubuntu 18 are not supported?

Big thanks.

umarov, 2020-02-03:
Yeah I wasn't able to install it using the directions for Ubuntu. With a live usb, I had to clone it, install `make` & `gcc`, then secure erase the nvme.

hreinecke, 2022-01-16:
Not much we can do. Please talk to the Ubuntu folks to fix up their release.

closed 2022-01-16 by hreinecke

igaw, 2022-01-17:
BTW, the official Ubuntu repository ships a packaged nvme-cli package. As far I remember the mentioned ppa is also pretty outdated.