linux-nvme/nvme-cli #659: NVMe discover failed with Input/output error with NVMe-cli ver 1.8.1

govind-rathod1987, 2020-02-13:
NVMe discover failed with Input/output error a below:

linux-q9xz:~ # nvme discover -t rdma -a 100.100.20.32
Failed to write to /dev/nvme-fabrics: Input/output error
linux-q9xz:~ #

dmesg output is:
[ 5743.295642] nvme nvme9: Connect command failed, error wo/DNR bit: 6
[ 5743.296728] nvme nvme9: failed to connect queue: 0 ret=6

using latest 5.3 kernel:
linux-q9xz:~ # uname -r
5.5.3-195-default
linux-q9xz:~ #

loaded Inbox drivers:
linux-q9xz:~ # lsmod | grep nvme
nvme_rdma              36864  0
rdma_cm                73728  1 nvme_rdma
ib_core               372736  6 rdma_cm,nvme_rdma,iw_cm,ib_uverbs,mlx5_ib,ib_cm
nvmet_fc               32768  1 lpfc
nvmet                  98304  1 nvmet_fc
configfs               53248  3 rdma_cm,nvmet
nvme_fc                45056  1 lpfc
nvme_fabrics           28672  2 nvme_fc,nvme_rdma
linux-q9xz:~ #

keithbusch, 2020-02-13:
This says you're getting a response from the target, and the response is '6', or Internal Error. That doesn't sound good at all. Is this a Linux target?

govind-rathod1987, 2020-02-19:
yes it is linux spdk target. it works with NVMe hosts with kernel ver 4.19 or 4.20 but not with NVMe host with kernel ver 5.3 onwards

keithbusch, 2020-02-19:
Sounds like an issue that should be brought to the kernel mailing list instead, linux-nvme@lists.infradead.org

keithbusch, 2020-02-20:
Or maybe open with spdk... From pursuing their code, it looks like their target would only return an internal error if it can't allocate something, like memory or unique identifiers. Anything else wrong in the host's command should have received a different error.

But the fact it worked before 5.3 (it works in 5.2?) does make linux kernel host seem suspicious, but it doesn't look like anything interesting changed there in 5.3.

govind-rathod1987, 2020-02-20:
Thanks for your comments, even i was suspecting the same.
I will close this case and open a new one against SPDK.

closed 2020-02-20 by govind-rathod1987