linux-nvme/nvme-cli #663: Build fails without libsystemd-dev package in Ubuntu

minwooim, 2020-02-18:
nvme-cli failed to build in fabrics.c due to 
```
#include <systemd/sd-id128.h>
```

```
minwoo@minwoo-desktop:~/work/nvme-cli.git$ (master) make
NVME_VERSION = 1.10.1.19.g0ff7a
    CC nvme-print.o
    CC nvme-ioctl.o
    CC nvme-lightnvm.o
    CC fabrics.o
fabrics.c:47:10: fatal error: systemd/sd-id128.h: No such file or directory
 #include <systemd/sd-id128.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:91: recipe for target 'fabrics.o' failed
make: *** [fabrics.o] Error 1
```

This is because lack of libsystem-dev package.  Should we have README for
prerequisites or some other way to avoid it?

keithbusch, 2020-02-18:
Hm, are we missing a package in our dependency checks or is it called something different in ubuntu? Works fine in fedora.

xahmad, 2020-02-29:
In Ubuntu package libsystem-dev is necessary. 

pkg-config --exists systemd  --atleast-version=232 returns 0 but headers are not available.

igaw, 2022-01-19:
This is fixed in master.

closed 2022-01-19 by igaw

mauricewells, 2022-11-30:
I resolved with the below command
`sudo apt install libsystemd-dev -y`