linux-nvme/nvme-cli #668: Building on FreeBSD fails

xahmad, 2020-02-25:
While there is a port available for the FreeBSD that install version 1.8.1 the code from this repo doesn't build on any version of the FreeBSD 12.1 / 13. 

Probably need to find out from what changes the port did to make it work & merge them to mainline.

darkfiberiru, 2020-12-18:
@bsdimp has branch here https://github.com/bsdimp/nvme-cli/commit/c92d8718bcd680587d3ee537220fb6a4d9f71107

bsdimp, 2020-12-18:
nmve-cli isn't well supported on FreeBSD. Chuck Cranor has done some work to make the device names more similar and provide a compat ioctl. But even that's just the start.

I pushed most of the portability changes already, but haven't had time to focus on finishing things beyond that because FreeBSD nvmecontrol meets my day-to-day needs.

keithbusch, 2020-12-19:
It could be shimmed with minor effort to get 90% function (sans idev and sysfs dependencies), but the interest in maintaining it isn't there. Native alternatives satisfy the immediate needs.

igaw, 2022-01-19:
For what's worth, with the split into nvme-cli and libnvme, the main bits to port are now in libnvme. But given there was almost no  progress here I am closing this issue. If there is still interest, please reopen.

closed 2022-01-19 by igaw