linux-nvme/nvme-cli #670: gp2 volume full

karthickrangan, 2020-02-29:
Hello friends,

In aws we are using gp2 storage volume and right now it was full

iam using git bash please tell a cmd to get into the volume and delete some files.

please do the needful
[image]

keithbusch, 2020-02-29:
Sounds like generic filesystem question and nothing to do with nvme specifically. It says the nvme partition is your root filesystem. What trouble are you having navigating from '/' and identifying undesirable files to delete?

karthickrangan, 2020-02-29:
Dear Keith,

i am using git bash tool to work with console
i need to delete some unwanted files in that root file system
what was the git bash cmd used to do that

karthickrangan, 2020-02-29:
awaithing for yo reply keith

keithbusch, 2020-02-29:
Sorry, your question has nothing to do with this project. I never heard of 'git bash'. It looks like a Windows tool, so this isn't the appropriate forums.

closed 2020-02-29 by keithbusch