linux-nvme/nvme-cli #678: Format worked with 1.9, but fails with 1.10.1

PatrickVerner, 2020-03-11:
I have a user that formatted his Samsung SSD 970 EVO Plus 250GB with 1.9, but the same drive will not format with 1.10.1. If he goes back to 1.9, the drive formats again. I personally don't own that drive, but the format works on my Samsung SSD 950 PRO 256GB with any version of nvme-cli. Not sure what changed that would cause this.

keithbusch, 2020-03-11:
I need to see the actual command that was attempted.

PatrickVerner, 2020-03-11:
nvme format /dev/nvme0n1

keithbusch, 2020-03-11:
If that's failing, it means the controller reports a format affects more than one namespace. This device has only one namespace though, so that's a weird thing for the controller to report.

Anyway, you need to specify the LBA format for such devices, like '`-l 0`', or whatever format index you're going for.

PatrickVerner, 2020-03-11:
> Anyway, you need to specify the LBA format for such devices, like '-l 0', or whatever format index you're going for.

Was this a change between 1.9 and 1.10.1? Because 1.9 works just fine for him with the same commands.

keithbusch, 2020-03-11:
Yes. Many people reported unintended data loss. We can't let that happen.

PatrickVerner, 2020-03-11:
Why does 1.10.1 format my drive and then most people's drives with nvme format /dev/nvme0n1?

I have exactly one user reporting that 1.10.1 does not work on the Samsung SSD 970 EVO Plus 250GB.

I guess I am confused. So moving forward what command should I use? This is all undocumented and your current documentation says that "nvme format /dev/nvme0n1" formats a drive with the default settings.

keithbusch, 2020-03-11:
nvme format /dev/nvme0n1 works as long as the controller reports that no one else is affected.

You need to specify the target lbaf if the format affects more than one namespace. Your device is what reports it.

> Why does 1.10.1 format my drive and then most people's drives with nvme format /dev/nvme0n1?

Because your device says a format is limited to that single namespace.

> I have exactly one user reporting that 1.10.1 does not work on the Samsung SSD 970 EVO Plus 250GB.
>
> I guess I am confused. So moving forward what command should I use? This is all undocumented and [...]

That command works if the affects of the format are contained to the targeted namespace. If the controller tells us that something else is affected by this command, you need to tell the tool which format you're selecting.

closed 2022-01-19 by igaw