linux-nvme/nvme-cli #688: nvme-read corrupts malloc buffer for metadata

dougmill-ibm, 2020-03-31:
The man page for nvme-read is vague about the -y (and also -z) options for specifying the metadata (and data) size. Because of this, an inexperienced user may use values that are too small. The code in submit_io() handles the case where too small of a size is given for -z, and ensures that the data buffer is the correct size. But in the case of the -y option, specifying too small of a size causes the read operation to overrun the buffer and corrupt the malloc arena. Example:
```
# nvme read /dev/nvme0n1 -s 0 -c 10 -z 4096 -y 8 -d data-1 -M meta-1
Rounding data size to fit block count (45056 bytes)
read: Success 
corrupted size vs. prev_size while consolidating
Aborted (core dumped) 
#
```
The -y option value needs to do something similar to -z in order to protect against the malloc corruption. In addition, it would be good to make the man page clearer about the values expected by -z and -y.

keithbusch, 2020-04-01:
point taken, we'll have to query the namespace identification to validate the user input; there's no other ways to get metadata sizes like there are for physical block sizes.

ganaidu707, 2021-02-07:
Hi @keithbusch , I have given a PR on this please, https://github.com/linux-nvme/nvme-cli/pull/898

igaw, 2022-01-19:
PR has been merged.