linux-nvme/nvme-cli #691: Add persistent event log from NVMe Spec 1.4

gcoakes, 2020-04-10:
[NVMe Spec 1.4](https://nvmexpress.org/wp-content/uploads/NVM-Express-1_4-2019.06.10-Ratified.pdf) defines a new log page which allows access to a record of significant events. In specific:

> The  Persistent  Event  Log  page  contains  information  about  significant  events  not  specific  to  a  particular  command.  The  information  in  this  log  page  shall  be  retained  across  power  cycles  and  resets

I'd like to discuss the feasibility of implementing this, the schema of the JSON output, and the format of the normal output.

keithbusch, 2020-04-14:
Correct, we don't have command-line level support for persistent event logs at the moment. This is actually quite a complicated log as it has multiple subtypes for each event, so implementing this feature would be a bit larger than every other nvme log.

I haven't got a device that supports this log, so haven't taken a stab at implementing support for it, but I'd be happy to take a pull request if someone can test their implementation.

Okisa123, 2020-05-20:
Keith, agreed with all above, it is not the same as telemetry logs 7 and 8 but log 7 information can be expanded in a persistent log. Do you believe it can be a case? Thanks in advance.

keithbusch, 2020-05-20:
There is a telemetry creation event in the persistent event log, but it contains only a snippet of what the telemetry log contains. But that's just one of 15 spec defined events contained in a persistent event log.

Okisa123, 2020-05-21:
Does Intel support persistent log or not yet?

Okisa123, 2020-05-21:
Thank you, Keith. It helps to see that telemetry logs itself must be done well. Being persistent is a separate task.

keithbusch, 2020-05-21:
> Does Intel support persistent log or not yet?

I wouldn't know.

keithbusch, 2020-05-21:
> Thank you, Keith. It helps to see that telemetry logs itself must be done well. Being persistent is a separate task.

The persistent event log is orthogonal to telemetry, really. The PEL is just saving events that have occured. Whether those events also correspond to another log that is persistent or not is not what PEL is about.

Specifically on telemetry persistence (which is not to be confused with the PEL), the spec says "The Telemetry Controller-Initiated Data shall persist across all resets. "

ganaidu707, 2021-01-11:
@keithbusch I have submitted PEL PR as NVMe1.4 Spec. 
https://github.com/linux-nvme/nvme-cli/pull/857

keithbusch, 2021-01-11:
Thanks, will take a look.

igaw, 2022-01-19:
PR has been merged.

closed 2022-01-19 by igaw