linux-nvme/nvme-cli #697: About nvme command return code

seanmaster, 2020-04-20:
[image]

Hi all, I use the latest nvme-cli version-V1.10.1 I found some different with nvme-cli-V1.5.x
I hope can got the real nvme return code after command by `echo $?` but the V1.10.1 I got the return code is remapping from errno.h.
For this change, Is there any way to get the return code more convenient? Or I only can get it from the parser the stdout message. Thanks~

keithbusch, 2020-04-20:
Yeah, nvme status codes are 15 bits, but shell return codes are only 8. Since we can't reliably return an nvme status code, and non-nvme related errors also return errno's that we couldn't distinguish from an nvme error, someone had made the change makes the program always exit with 0 or errno.

seanmaster, 2020-05-25:
Hi Keithbusch,
Thanks your response, so in other world, we only can parser the nvme return code by stdout/stderr string, if cmd return code is not equal 0.

igaw, 2022-01-19:
@hreinecke this reminds me that we increased the error range with 31c533d50b02 ("log: Introduce libnvme error codes") in libnvme. We should probably filter those out and just exit with 1.

igaw, 2022-01-19:
And btw, I think we should return 1 in all error cases.

hreinecke, 2022-01-20:
'-1', please. '1' is daft, as one is left to wonder if '1' is a success (as would be if interpreted as 'bool') or a failure. If you want to stay with '1' we should mark the functions as 'bool', not 'int'.

hreinecke, 2022-01-20:
I would stay with the libc convention of returning '-1' and setting 'errno' for all API functions. Internal functions can return the errno as a negative value or '0' on success.

igaw, 2022-01-20:
-1 is really not a great choice.

The `exit` function (and also `int main() { return err; }` ) use an `int`, but only the lower 8 bits are actually part of the 'status' (see status argument of `wait` and `waitpid`) . 

There is no fixed rule that is correct. SUSv3 specifies EXIT_SUCCESS(0) and EXIT_FAILURE(1).

And one thing to remember is that values above 128 can cause confusion in shell scripts. The reason is that, when a command is terminated by a signal, the shell indicates this fact by setting the value of the variable `$?` to 128 plus the signal number.

PR #1345 (2022-01-29): Handle libnvme error codes and return proper error code when termiating nvme-cli
Fixes #697
closed 2022-01-29 by hreinecke