linux-nvme/nvme-cli #698: Version 1.11 causes warnings during the build

primeos, 2020-04-24:
Quick FYI: Since the Makefile defaults to `-Wall` and `-Werror` the build for Version 1.11 fails with:
```
    CC nvme-topology.o
nvme-topology.c: In function 'verify_legacy_ns':
nvme-topology.c:361:32: error: format not a string literal and no format arguments [-Werror=format-security]
  361 |    asprintf(&n->ctrl->address, tmp_address);
      |                                ^~~~~~~~~~~
nvme-topology.c:360:4: error: ignoring return value of 'asprintf', declared with attribute warn_unused_result [-Werror=unused-result]
  360 |    asprintf(&n->ctrl->transport, "pcie");
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nvme-topology.c:361:4: error: ignoring return value of 'asprintf', declared with attribute warn_unused_result [-Werror=unused-result]
  361 |    asprintf(&n->ctrl->address, tmp_address);
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:96: nvme-topology.o] Error 1
```

Tested with GCC 9.3.0.
The code from the warnings above was introduced in b9b16d5170f23433a30cb88eefd2ab30b48d151e.

primeos, 2020-04-24:
Closing this as it is already fixed by a032a481d7406c5a0ae188eeaae709805d1c6e02 in version 1.11.1.

closed 2020-04-24 by primeos

keithbusch, 2020-04-24:
thanks for the breakage notice. i was under some time pressure to get a release out and didn't build this in enough environments, so i botched that one. Thanks to Chaitanya for the quick fix.