linux-nvme/nvme-cli #699: Add CI to automatically test the builds?

primeos, 2020-04-24:
After #698 I noticed that there don't seem to be any automatic builds, at least not with GitHub integration. It might be sense to run some automatic builds, ideally also with different compilers and versions. This could e.g. be done via GitHub Actions or Travis CI.

Would that be welcome? If it's just a matter of time I might be able to give it a shot (though I cannot guarantee I'll have enough time atm).

sc108-lee, 2020-07-03:
I think you can close this issue, since @keithbusch did the job for this.
https://github.com/linux-nvme/nvme-cli/blob/master/.github/workflows/c-cpp.yml

primeos, 2020-07-03:
Awesome, thanks for letting me know :)

In case there will be problems due to different compilers (e.g. Clang) or compiler versions it might make sense to configure a build matrix as well:
- https://docs.github.com/en/actions/configuring-and-managing-workflows/configuring-a-workflow#configuring-a-build-matrix
- https://stackoverflow.com/questions/57810623/how-to-select-the-c-c-compiler-used-for-a-github-actions-job/57814146#57814146

But I guess the current implementation should be fine for now.

closed 2020-07-03 by primeos