linux-nvme/nvme-cli #703: Read Nlog in nvme cli fails due to invalid field

Sumithra92, 2020-04-30:
Commands in nvme cli to read internal NLogs (Tried with v1.8/v1.9/v1.11.1): 
**nvme intel internal-log /dev/nvme0 -l 0 -r 1 -m 1**

The above command fails with the following error message:

failed on cmd.data_len 4096 cmd.cdw13 0 cmd.cdw12 28 cmd.cdw10 1024 err 4002 remaining size 1024
--
NVMe Status:INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(4002)

keithbusch, 2020-04-30:
does your drive actually support this vendor specific log page?

Sumithra92, 2020-04-30:
> does your drive actually support this vendor specific log page?

Hi Keith - Yes. The drive supports nlogs.

Sumithra92, 2020-06-03:
Hi! 
I do apologize for the long inactivity. We figured out the issue for now and are making necessary changes to the product. This issue can be closed for now.

closed 2020-06-03 by Sumithra92