linux-nvme/nvme-cli #709: After format re-read partition failed

sc108-lee, 2020-05-15:
since checking block device is removed by below commit.
https://github.com/linux-nvme/nvme-cli/commit/e770466615096a6d41f038a28819b00bc3078e1d#diff-a71a06d2f8c91ea6e57b2f28e8813cfdR2862

There is always fail for format with character device. 
like nvme /dev/nvme0
case 1. nvme /dev/nvme0 -n 0xffffffff (solved)
case 2. nvme /dev/nvme0 -n 1 ( prev lbaf == new lbaf ) (solved)
    case 1, 2 solved by #708 
    https://github.com/linux-nvme/nvme-cli/pull/708/commits/e9e0de444b76cea8133b42b62192ea99d81e7515

**case 3: nvme /dev/nvme0 -n 1 ( prev lbaf != new lbaf )**
always fail to issue ioctl(fd, BLKRRPART) on character device.
Do we need this failure checking???
If the device AER enabled by set-feature, it would be handled by kernel.

I would love to hear opinion @tkulasek @minwooim

keithbusch, 2020-05-15:
We'd also need the device to actually send the AER for the kernel to react, which some do, some don't. Instead of the RRPART ioctl, nvme-cli should send NVME_IOCTL_RESCAN.

sc108-lee, 2020-05-18:
hummm.
@keithbusch I agree to send NVME_IOCTL_RESCAN for character device with ALL NS.
But for case 3, It would be different with intention. ( target ns 1, but rescan all )
Not sure this is right way to do this.

And also for spdk device will be a character device.
would be fine to do NVME_IOCTL_RESCAN for that?

keithbusch, 2020-05-18:
There's no harm in having the kernel revalidate all the disks for this exceptional condition.

No clue about spdk's handling.

sc108-lee, 2020-05-19:
@keithbusch thanks for confirming this.
since there is no harm for this, I raise the PR for this.
https://github.com/linux-nvme/nvme-cli/pull/712

tkulasek, 2020-05-20:
SPDK operates in the user space and extending to handle additional ioctls should be not the problem in the future.

closed 2020-07-14 by sc108-lee