linux-nvme/nvme-cli #710: get-log -o

Baaraan, 2020-05-16:
Hello,
I'm using nvme-cli version 0.5-1ubuntu0.2 (just updated) and when I try to read a specific offset of a log-page, I get this error:

```
get-log: unrecognized option '-o'
Usage: nvme get-log <device> [OPTIONS]

Retrieve desired number of bytes from a given log on a specified device in
either hex-dump (default) or binary format

Options:
  [  --namespace-id=<NUM>, -n <NUM> ]   --- desired namespace
  [  --log-id=<NUM>, -i <NUM> ]         --- identifier of log to retrieve
  [  --log-len=<NUM>, -l <NUM> ]        --- how many bytes to retrieve
  [  --raw-binary, -b ]                 --- output in raw format
```
Does that mean -o being removed, is that intentionally?

Thanks,
Maliheh

keithbusch, 2020-05-16:
Not removed. The version you are using was released before log page offsets existed.

Baaraan, 2020-05-18:
Thanks Keith, 
I was using the one built-in my Ubuntu and sure it was old.

closed 2020-05-18 by Baaraan

Baaraan, 2020-05-19:
Hi Keith,

Now that I've actually updated, get-log is all good,  but I have another problem with this new version:
nvme version 1.11.1.17.g46e0
Problem is nvme list does not show my nvme drive anymore while I'm pretty sure my drive is there happy and healthy:

`sudo lsblk`
`nvme0n1                       259:0    0     7T  0 disk`

`sudo nvme list`
`Failed to scan namespaces`

Any idea what I might be doing wrong or maybe someone else also has same problem with this version?

Thanks!

keithbusch, 2020-05-19:
weird, what kernel version are you using?

Baaraan, 2020-05-19:
Here:

4.10.0-28-generic #32~16.04.2-Ubuntu SMP Thu Jul 20 10:19:48 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

Thanks for your time!

Baaraan, 2020-05-19:
@keithbusch ,
FYI I've went back to  1.10.1 tag and now everything seems working again, so sure something in the latest, if you want I can make a new issue for this to be tracked?

keithbusch, 2020-05-20:
Could you try pulling the latest and see if it works? Commit https://github.com/linux-nvme/nvme-cli/commit/f4f3fd2ea4b878fe38fd3409f374ded8d5b34a4c was just added that should address your 'list' sighting.

Baaraan, 2020-05-21:
Yes, all good now :)
Thanks for the quick fix!