linux-nvme/nvme-cli #716: make fails in wdc_do_cap_dui plugin

narasimhan-v, 2020-05-21:
Hi,

This is on a Redhat Linux system

```
# make
NVME_VERSION = 1.11.1.21.gd766
    CC nvme-print.o
    CC nvme-ioctl.o
    CC nvme-lightnvm.o
    CC fabrics.o
    CC nvme-models.o
    CC plugin.o
    CC nvme-status.o
    CC nvme-filters.o
    CC nvme-topology.o
    CC plugins/intel/intel-nvme.o
    CC plugins/lnvm/lnvm-nvme.o
    CC plugins/memblaze/memblaze-nvme.o
    CC plugins/wdc/wdc-nvme.o
plugins/wdc/wdc-nvme.c: In function ‘wdc_do_cap_dui’:
plugins/wdc/wdc-nvme.c:1673:51: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 4 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
     fprintf(stderr, "%s: INFO : WDC : Offset 0x%llx exceeds total size 0x%llx, no data retrieved\n",
                                                ~~~^
                                                %lx
      __func__, offset, total_size);
                ~~~~~~                              
plugins/wdc/wdc-nvme.c:1673:77: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 5 has type ‘__s64’ {aka ‘long int’} [-Werror=format=]
     fprintf(stderr, "%s: INFO : WDC : Offset 0x%llx exceeds total size 0x%llx, no data retrieved\n",
                                                                          ~~~^
                                                                          %lx
      __func__, offset, total_size);
                        ~~~~~~~~~~                                            
plugins/wdc/wdc-nvme.c:1705:52: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 4 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
      fprintf(stderr, "%s: INFO : WDC : Offset 0x%llx, file size 0x%llx, total size 0x%llx, log size 0x%llx\n",
                                                 ~~~^
                                                 %lx
       __func__, offset, file_size, total_size, log_size);
                 ~~~~~~                              
plugins/wdc/wdc-nvme.c:1705:70: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 5 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
      fprintf(stderr, "%s: INFO : WDC : Offset 0x%llx, file size 0x%llx, total size 0x%llx, log size 0x%llx\n",
                                                                   ~~~^
                                                                   %lx
       __func__, offset, file_size, total_size, log_size);
                         ~~~~~~~~~                                     
plugins/wdc/wdc-nvme.c:1705:89: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 6 has type ‘__s64’ {aka ‘long int’} [-Werror=format=]
      fprintf(stderr, "%s: INFO : WDC : Offset 0x%llx, file size 0x%llx, total size 0x%llx, log size 0x%llx\n",
                                                                                      ~~~^
                                                                                      %lx
       __func__, offset, file_size, total_size, log_size);
                                    ~~~~~~~~~~                                            
plugins/wdc/wdc-nvme.c:1705:106: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 7 has type ‘__s64’ {aka ‘long int’} [-Werror=format=]
      fprintf(stderr, "%s: INFO : WDC : Offset 0x%llx, file size 0x%llx, total size 0x%llx, log size 0x%llx\n",
                                                                                                       ~~~^
                                                                                                       %lx
       __func__, offset, file_size, total_size, log_size);
                                                ~~~~~~~~                                                   
cc1: all warnings being treated as errors
make: *** [Makefile:93: plugins/wdc/wdc-nvme.o] Error 1

```

narasimhan-v, 2020-05-21:
Preventing the plugin from installing, by commenting out the line 
`        plugins/wdc/wdc-nvme.o                  \`
proceeds with make successful

keithbusch, 2020-05-21:
32-bit? Pushed appropriate fixes.

narasimhan-v, 2020-05-21:
It is a 64-bit power pc little endian

narasimhan-v, 2020-05-21:
@keithbusch Thanks, https://github.com/linux-nvme/nvme-cli/commit/73a04020ff8dc5399bb47ef16047902f2b63bd66 fixes the issue

closed 2020-05-21 by narasimhan-v