linux-nvme/nvme-cli #723: 'list' command fails for v1.11.1; works on HEAD or v.1.10.1

mikepyne, 2020-05-27:
Running `nvme list` fails in version 1.11.1, and works in version 1.10.1 and HEAD (593e504)

I cloned the repository to a new folder, then changed into the folder and built the utility.

    $ ./nvme version
    nvme version 1.11.1.26.g593e

    $ ./nvme list
    Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
    ---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
    /dev/nvme0n1     38JS102PT25Q         TOSHIBA CONTROLLER MODEL NUMBER SAMPLE   1           8.34  GB /   1.92  TB      4 KiB +  0 B   1CET6105
    /dev/nvme1n1     88AF7041FP9P         KXG6AZNV512G TOSHIBA                     1         512.11  GB / 512.11  GB    512   B +  0 B   AGFE5101

Checked out version 1.11.1, using `git checkout v1.11.1`, then ran `make clean && make`.

    $ ./nvme version
    nvme version 1.11.1

    $ ./nvme list
    Failed to scan namespaces

Checkout version 1.10.1 (`git checkout v1.10.1`), ran `make clean && make`.

    $ ./nvme version
    nvme version 1.10.1

    $ ./nvme list
    Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
    ---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
    /dev/nvme0n1     38JS102PT25Q         TOSHIBA CONTROLLER MODEL NUMBER SAMPLE   1           8.34  GB /   1.92  TB      4 KiB +  0 B   1CET6105
    /dev/nvme1n1     88AF7041FP9P         KXG6AZNV512G TOSHIBA                     1         512.11  GB / 512.11  GB    512   B +  0 B   AGFE5101

If it is of interest, I'm running on CentOS 7 (`Linux 4.4.212-1.el7.elrepo.x86_64 x86_64`).

Obviously, whatever broke in 1.11.1 has already been fixed (not that I could figure out what had changed!), However, a tagged release apparently has a broken command, so worth considering another patch release?

keithbusch, 2020-05-27:
Thanks for the notice. The issue should be fixed with commit https://github.com/linux-nvme/nvme-cli/commit/f4f3fd2ea4b878fe38fd3409f374ded8d5b34a4c.

keithbusch, 2020-05-27:
I'll port it to a branch specific to the tagged release and spin a new version.

keithbusch, 2020-05-27:
v1.11.2 tagged with the single fix

mikepyne, 2020-05-28:
I've built the new tag and the `list` command works correctly.

Thanks!

closed 2020-05-28 by mikepyne