linux-nvme/nvme-cli #728: [build failure] wdc plugin

tim-oleksii, 2020-05-28:
I'd really love to fix this bug but nothing else then creating EOL_STATE_UNKNOWN is not available for me

gcc 10.1

```
[tim-oleksii@archlinux linux_nvme]$ git clone https://github.com/linux-nvme/nvme-cli.git .
Cloning into '.'...
remote: Enumerating objects: 33, done.
remote: Counting objects: 100% (33/33), done.
remote: Compressing objects: 100% (28/28), done.
remote: Total 9372 (delta 11), reused 20 (delta 5), pack-reused 9339
Receiving objects: 100% (9372/9372), 4.43 MiB | 4.48 MiB/s, done.
Resolving deltas: 100% (7085/7085), done.
[tim-oleksii@archlinux linux_nvme]$ make nvme
NVME_VERSION = 1.11.1.28.gf852
    CC nvme-print.o
    CC nvme-ioctl.o
    CC nvme-lightnvm.o
    CC fabrics.o
    CC nvme-models.o
    CC plugin.o
    CC nvme-status.o
    CC nvme-filters.o
    CC nvme-topology.o
    CC plugins/intel/intel-nvme.o
    CC plugins/lnvm/lnvm-nvme.o
    CC plugins/memblaze/memblaze-nvme.o
    CC plugins/wdc/wdc-nvme.o
plugins/wdc/wdc-nvme.c: In function ‘wdc_drive_status’:
plugins/wdc/wdc-nvme.c:3932:3: error: ‘system_eol_state’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
 3932 |   printf("  Drive Life Status:   Unknown : 0x%08x/0x%08x\n",
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 3933 |     le32_to_cpu(user_eol_state), le32_to_cpu(system_eol_state));
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
plugins/wdc/wdc-nvme.c:3932:3: error: ‘user_eol_state’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
cc1: all warnings being treated as errors
make: *** [Makefile:94: plugins/wdc/wdc-nvme.o] Error 1
[tim-oleksii@archlinux nvme-cli]$ gcc --version
gcc (GCC) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
[tim-oleksii@archlinux nvme-cli]$ uname -a
Linux archlinux 5.6.13-arch1-1 #1 SMP PREEMPT Thu, 14 May 2020 06:52:53 +0000 x86_64 GNU/Linux

```

keithbusch, 2020-05-28:
Hm, gcc 9.2.1 doesn't complain, but it does appear to be a legit warning. Fix pushed.

tim-oleksii, 2020-05-28:
Thx for a fix

closed 2020-05-28 by tim-oleksii