linux-nvme/nvme-cli #731: WDC plugin error: "WDC: unsupported device for this command"

rugk, 2020-06-07:
The WDC plugin does not seem to detect my drive as an NVMe drive.

## STR

```
# nvme wdc get-drive-status /dev/nvme0
ERROR : WDC: unsupported device for this command
# nvme wdc cap-diag /dev/nvme0
ERROR : WDC: unsupported device for this command
# nvme wdc drive-log /dev/nvme0
ERROR : WDC: unsupported device for this command
# nvme wdc get-crash-dump /dev/nvme0
ERROR : WDC: unsupported device for this command
```

Also, some other commands return strange results IMHO (I'm no expert in this hardware stuff.):
```
# nvme device-self-test /dev/nvme0
NVMe status: INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x4002)
# nvme get-log /dev/nvme0
Invalid log identifier: -1. Valid range: 0-255
# # nvme telemetry-log /dev/nvme0 # returns a binary file > 30 kB
```

## System

```
# nvme version
nvme version 1.10.1
# nvme wdc version
nvme wdc version 1.10.1
# nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     20070X442812         WDC WDS100T2B0C-00PXH0                   1           1,00  TB /   1,00  TB    512   B +  0 B   211070WD
```

### OS

Fedora 32 Silverblue (v 32.20200605.0)
x64
GNOME v3.36.2
Wayland  
Linux kernel: Linux ***** 5.6.15-300.fc32.x86_64 #1 SMP Fri May 29 14:23:59 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux  

### Hardware

device: IdeaPad 5 15ARE05 [81YQ (LENOVO_MT_81YQ_BU_idea_FM_IdeaPad 5 15ARE05)]
CPU: AMD Ryzen 7 4800U with radeon graphics x16
NVMe: WDC WDS100T2B0C-00PXH0

**Complete log/command output:** (some general things + lshw + nvme-cli)
https://gist.github.com/rugk/8c87d8cd8b580c286f51eb243d889f50

keithbusch, 2020-06-07:
> The WDC plugin does not seem to detect my drive as an NVMe drive.
>
> ## STR
> ```
> # nvme wdc get-drive-status /dev/nvme0
> ERROR : WDC: unsupported device for this command
> # nvme wdc cap-diag /dev/nvme0 [...]

At least according to the compatible list that was written by the WDC developers, the device you're using doesn't support the commands you're trying to run.

> Also, some other commands return strange results IMHO (I'm no expert in this hardware stuff.):
> 
> ```
> # nvme device-self-test /dev/nvme0
> NVMe status: INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x4002)

The device does not support the optional self-test command.

> # nvme get-log /dev/nvme0
> Invalid log identifier: -1. Valid range: 0-255

You didn't specify a log to retrieve.

> # # nvme telemetry-log /dev/nvme0 # returns a binary file > 30 kB

Telemetry logs are vendor specific. Most vendors implement these as a binary file that only the vendor can decode.

rugk, 2020-06-07:
Uff… so WDC commands are not supported by all WDC devices, that's awkward. MAybe you can at least improve the error message to make that obvious?

keithbusch, 2020-06-08:
@rbates0119, could you take a look?

closed 2022-01-19 by igaw

rugk, 2022-01-19:
Well the last thing stated was:
> Maybe you can at least improve the error message to make that obvious?

keithbusch, 2022-01-19:
> Well the last thing stated was:
> 
> > Maybe you can at least improve the error message to make that obvious?

Do you have a suggestion? The existing error looks pretty obvious that the requested operation is not supported by the device.

CyberNitta, 2022-02-05:
same problem with this device,,,

rugk, 2022-02-05:
Hmm, I don't recall it exactly, but from what I read I was surprised that if I use the WDC command explicitly, not all WDC devies support all WDC commands.

So the error was:
> WDC: unsupported device for this command

A better error could be:
> WDC: this WDC device does not support this command

This makes it sure that it is not on me using the wrong command. Just an idea though…

igaw, 2022-02-07:
This is a question/idea for @jeffreyalien

jeffreyalien, 2022-02-07:
I can look into changing the messages to be a little more clear.  I probably won't be able to get at this right away though.