linux-nvme/nvme-cli #740: Ignored some values of ctrl_loss_tmo and keep_alive_tmo

alekseymmm, 2020-06-16:
In fabrics.c there are checks for the correct values of `ctrl_loss_tmo` and `keep_alive_tmo` on `nvme connect` command. These checks look like 
```
if (cfg.ctrl_loss_tmo) {
...
}
```
and 
```
if (cfg.keep_alive_tmo) {
...
}
```
But according to parsing of these parameters in kernel (look at https://elixir.bootlin.com/linux/latest/C/ident/nvmf_parse_options) in switch cases for 
```
		case NVMF_OPT_KATO:
			if (match_int(args, &token)) {
				ret = -EINVAL;
				goto out;
			}

			if (token < 0) {
				pr_err("Invalid keep_alive_tmo %d\n", token);
				ret = -EINVAL;
				goto out;
``` 
So the 0 value of `keep_alive_tmo`  is considered to be correct by the kernel driver. And
```
		case NVMF_OPT_CTRL_LOSS_TMO:
			if (match_int(args, &token)) {
				ret = -EINVAL;
				goto out;
			}

			if (token < 0)
				pr_warn("ctrl_loss_tmo < 0 will reconnect forever\n");
			ctrl_loss_tmo = token;
			break;
```
All values of `ctrl_loss_tmo` even 0 and negative are correct but have special handling.

If I understand params handling in nvme-cli correctly then probably this behavior could be fixed by a small patch like this
```
# git diff
diff --git a/fabrics.c b/fabrics.c
index 2fec802..ba62110 100644
--- a/fabrics.c
+++ b/fabrics.c
@@ -845,11 +845,11 @@ static int build_options(char *argstr, int max_len, bool discover)
                                cfg.queue_size, false)) ||
            (!discover &&
              add_int_argument(&argstr, &max_len, "keep_alive_tmo",
-                               cfg.keep_alive_tmo, false)) ||
+                               cfg.keep_alive_tmo, true)) ||
            add_int_argument(&argstr, &max_len, "reconnect_delay",
                                cfg.reconnect_delay, false) ||
            add_int_argument(&argstr, &max_len, "ctrl_loss_tmo",
-                               cfg.ctrl_loss_tmo, false) ||
+                               cfg.ctrl_loss_tmo, true) ||
            add_int_argument(&argstr, &max_len, "tos",
                                cfg.tos, true) ||
            add_bool_argument(&argstr, &max_len, "duplicate_connect",
@@ -1003,7 +1003,7 @@ retry:
                p += len;
        }
 
-       if (cfg.ctrl_loss_tmo) {
+       if (cfg.ctrl_loss_tmo >= -1) {
                len = sprintf(p, ",ctrl_loss_tmo=%d", cfg.ctrl_loss_tmo);
                if (len < 0)
                        return -EINVAL;
@@ -1017,7 +1017,7 @@ retry:
                p += len;
        }
 
-       if (cfg.keep_alive_tmo) {
+       if (cfg.keep_alive_tmo >= 0) {
                len = sprintf(p, ",keep_alive_tmo=%d", cfg.keep_alive_tmo);
                if (len < 0)
                        return -EINVAL;
```

Let me know if I understand the current behavior correctly, and whether the patch is correct and doesn't miss anything.

keithbusch, 2020-06-16:
Could you send this to the kernel mailing list? The fabrics devs don't check this forum very often.

alekseymmm, 2020-06-16:
@keithbusch Hi, are you sure this possible bug is for kernel developers? In my view, it is entirely nvme-cli argument handling error. Or nvme-cli is maintained by kernel developers? if so then do you know what mailing list would be more appropriate regarding this matter?

keithbusch, 2020-06-16:
well, i'm just not sure if what the kernel is doing was intended, as the constraints set in the tooling prevent testing those kernel paths.

A lot of the contributions to this tool come from the kernel developers too

@sagigrimberg , do you know off the top of your head about this mismatch between kernel and user space, and if so, is the user space constraining these parameters unnecessarily?

sagi-lb, 2020-06-16:
This patch looks fine, I guess that keep_alive_tmo=0 and ctrl_loss_tmo=0 are not popular values
so we didn't hit this before. However, keep_alive_tmo is actually just a debug option to exercise the keep-alive timeout, not sure we want to expose it to nvme-cli users..

keithbusch, 2020-06-16:
cool, thanks for confirming.

@alekseymmm , do you want to provide a proper patch for this?

alekseymmm, 2020-06-18:
@sagi-lb Hello. After some testing, I confirm that using keep_alive_tmo=0 is not a good idea. Usually, the target fails to perform connect with this value. So maybe it is a good way of not letting set this value in nvme-cli.
But ctrl_loss_tmo=0 is quite useful for me and maybe others when we create a RAID with redundancy over multiple nvmeof targets. In case of some target disconnect, it is better to lose the device within a raid, quickly recover necessary data and return it to the user than wait for timeout of controller_loss hence heavily increasing the latency of those ios. 
I can do the patch for ctrl_loss_tmo value, what is the best way though? Make a PR here or just provide patch here and some of you guys will apply it?

keithbusch, 2020-06-18:
You may send the patch as either a pull request in the github interface, or email the patch to linux-nvme@lists.infradead.org.

alekseymmm, 2020-06-19:
@keithbusch @sagi-lb Thank you.

closed 2020-06-19 by alekseymmm