linux-nvme/nvme-cli #762: sizeof(struct nvme_iocs_vector

ygoldfill, 2020-06-26:


ygoldfill, 2020-06-26:
suggest a fix
diff --git a/linux/nvme.h b/linux/nvme.h
index 08d5603..b066532 100644
--- a/linux/nvme.h
+++ b/linux/nvme.h
@@ -424,8 +424,8 @@ struct nvme_iocs_vector {
                __le64 rsvd1 : 1;
                __le64 zoned : 1;
                __le64 rsvd3 : 61;
+               __le64 a;
        };
-       __le64 a;
 };

keithbusch, 2020-06-26:
Thanks for the catch. I can't believe I didn't see this -- it's using bitfields, and that's a big no-no for such a program where bit order actually matters.

keithbusch, 2020-06-26:
I fixed it up. There was more wrong with this that just the size of this field, should be okay now.

closed 2020-06-26 by keithbusch

ygoldfill, 2020-06-26:
Thanks for fixing
Shouldn't the print of this command be more descriptive as to pointing the NVM and ZONE like in
get feature 0x19 and nvme_show_iocs_vector ?

keithbusch, 2020-06-26:
That was broken, feature 19 doesn't return the bit mask of the supported command sets. It returns an index into the identify iocs, which is retrieved via a completely different command.

As far as decoding the iocs from that identification, yah, that will be added shortly.