linux-nvme/nvme-cli #772: Compiling failed on ppc64le

yizhanglinux, 2020-07-07:
Hi

I found this compiling issue on ppc64le, and cannot reproduced on x86_64.

```
# make
    CC plugins/scaleflux/sfx-nvme.o
plugins/scaleflux/sfx-nvme.c: In function ‘bd_table_show’:
plugins/scaleflux/sfx-nvme.c:533:17: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
   printf(" 0x%llx", *(bb_elem++));
              ~~~^   ~~~~~~~~~~~~
              %lx
plugins/scaleflux/sfx-nvme.c:541:17: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
   printf(" 0x%llx",*(bb_elem++));
              ~~~^  ~~~~~~~~~~~~
              %lx
plugins/scaleflux/sfx-nvme.c: In function ‘show_cap_info’:
plugins/scaleflux/sfx-nvme.c:600:49: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  printf("logic            capacity:%5lluGB(0x%llx)\n",
                                              ~~~^
                                              %lx
    IDEMA_CAP2GB(ctx->user_space), ctx->user_space);
                                   ~~~~~~~~~~~~~~~
plugins/scaleflux/sfx-nvme.c:602:49: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  printf("provisioned      capacity:%5lluGB(0x%llx)\n",
                                              ~~~^
                                              %lx
    IDEMA_CAP2GB(ctx->phy_space), ctx->phy_space);
                                  ~~~~~~~~~~~~~~  
plugins/scaleflux/sfx-nvme.c:604:49: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  printf("free provisioned capacity:%5lluGB(0x%llx)\n",
                                              ~~~^
                                              %lx
    IDEMA_CAP2GB(ctx->free_space), ctx->free_space);
                                   ~~~~~~~~~~~~~~~
plugins/scaleflux/sfx-nvme.c:606:49: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  printf("used provisioned capacity:%5lluGB(0x%llx)\n",
                                              ~~~^
                                              %lx
plugins/scaleflux/sfx-nvme.c:608:4:
    ctx->phy_space - ctx->free_space);
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~              
plugins/scaleflux/sfx-nvme.c: In function ‘change_sanity_check’:
plugins/scaleflux/sfx-nvme.c:697:5: error: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
     "WARNING: Free memory is not enough! "
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
plugins/scaleflux/sfx-nvme.c:700:5:
     mem_need, s_info.freeram);
     ~~~~~~~~
plugins/scaleflux/sfx-nvme.c:699:35: note: format string is defined here
     "WARNING: Memory needed is %llu, free memory is %lu\n",
                                ~~~^
                                %lu
plugins/scaleflux/sfx-nvme.c: In function ‘change_cap’:
plugins/scaleflux/sfx-nvme.c:777:17: error: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  printf("%dG %lluB %llu 4K\n",
              ~~~^
              %lu
   cfg.capacity_in_gb, cfg.cap_in_byte, cap_in_4k);
                       ~~~~~~~~~~~~~~~
plugins/scaleflux/sfx-nvme.c:777:23: error: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 4 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  printf("%dG %lluB %llu 4K\n",
                    ~~~^
                    %lu
   cfg.capacity_in_gb, cfg.cap_in_byte, cap_in_4k);
                                        ~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:95: plugins/scaleflux/sfx-nvme.o] Error 1
]# gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/ppc64le-redhat-linux/8/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: ppc64le-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-targets=powerpcle-linux --disable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --disable-libmpx --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --enable-secureplt --with-long-double-128 --with-cpu-32=power8 --with-tune-32=power8 --with-cpu-64=power8 --with-tune-64=power8 --build=ppc64le-redhat-linux
Thread model: posix
gcc version 8.3.1 20191121 (Red Hat 8.3.1-5) (GCC) 

```

keithbusch, 2020-07-07:
Ah crap, thanks for the head's up. Will get a fix asap and will see if i can add this type of check to the CI.

ericcurtin, 2021-02-02:
linked #784

ericcurtin, 2021-02-08:
This could be closed now I think @keithbusch

closed 2021-02-09 by keithbusch