linux-nvme/nvme-cli #775: libhugetlbfs : Warning error comes in firmware download command

visheshvijay, 2020-07-12:
Hi All,

From nvme cli 1.10 / 1.11/ 1.12  whenever I download the firmware binary I get below warning. 
_nvme fw-download /dev/nvme0 -f <fw_binary>
libhugetlbfs: WARNING: get_huge_pages: New region mapping failed (flags: 0x1): Cannot allocate memory
Firmware download success_

**Firmware download is success full but I want to know why this error comes and how can i rid of it.**

When I tried with nvme cli 1.9 then this error does not come.
OS - CentOS 7.4

Thanks,
Vishesh

visheshvijay, 2020-07-16:
Hi @keithbusch ,

Any Idea how can we avoid this error. Any seeting needs to be done on linux side.

keithbusch, 2020-07-16:
It just means there wasn't a hugepage to allocate and the library's debug verbosity is set high enough to emit the warning. It's otherwise harmless, but I'll get it suppressed.

closed 2020-07-16 by keithbusch

visheshvijay, 2020-07-17:
@keithbusch  Thanks a lot.
Its working fine now.