linux-nvme/nvme-cli #779: Trying to format a locked ssd using nvme-cli gives INVALID_OPCODE error

deepakagrawal, 2020-07-18:
Hi I have a Samsung 970 EVO plus which I locked from bios. I tried recovering the data and I couldn't so I am just formatting the drive so that I can use it again. I am using the `nvme format` command to format the drive. But it gives me the following error.

```
    root@deepak-GL502VM:/home/deepak# nvme format /dev/nvme0 -n 1
    NVMe status: INVALID_OPCODE: The associated command opcode field is not valid(0x1)

```

Any help would be greatly appreciated.

keithbusch, 2020-07-18:
Is this a duplicate of https://github.com/linux-nvme/nvme-cli/issues/627
?

Flameancer, 2020-08-21:
It looks like this is still an issue. `Systemctl suspend` Does not appear to fix.

closed 2022-01-19 by igaw