linux-nvme/nvme-cli #784: fails to compile on ppc64el

anonymouse64, 2020-07-27:
On ubuntu 20.04 ppc64el, using master @ https://github.com/linux-nvme/nvme-cli/commit/8d39b9ba2ba815ea32a3d034a2f6201d43b49522 I see the following errors compiling:

```
+ make -j4
NVME_VERSION = 1.12.79.g8d39b
    CC nvme-print.o
    CC nvme-ioctl.o
    CC nvme-lightnvm.o
    CC fabrics.o
    CC nvme-models.o
    CC plugin.o
    CC nvme-status.o
    CC nvme-filters.o
    CC nvme-topology.o
    CC plugins/intel/intel-nvme.o
    CC plugins/lnvm/lnvm-nvme.o
    CC plugins/memblaze/memblaze-nvme.o
    CC plugins/wdc/wdc-nvme.o
    CC plugins/wdc/wdc-utils.o
    CC plugins/huawei/huawei-nvme.o
    CC plugins/netapp/netapp-nvme.o
    CC plugins/toshiba/toshiba-nvme.o
    CC plugins/micron/micron-nvme.o
    CC plugins/seagate/seagate-nvme.o
    CC plugins/virtium/virtium-nvme.o
    CC plugins/shannon/shannon-nvme.o
    CC plugins/dera/dera-nvme.o
    CC plugins/scaleflux/sfx-nvme.o
plugins/scaleflux/sfx-nvme.c: In function ‘bd_table_show’:
plugins/scaleflux/sfx-nvme.c:533:17: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  533 |   printf(" 0x%llx", *(bb_elem++));
      |              ~~~^   ~~~~~~~~~~~~
      |                 |   |
      |                 |   __u64 {aka long unsigned int}
      |                 long long unsigned int
      |              %lx
plugins/scaleflux/sfx-nvme.c:541:17: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  541 |   printf(" 0x%llx",*(bb_elem++));
      |              ~~~^  ~~~~~~~~~~~~
      |                 |  |
      |                 |  __u64 {aka long unsigned int}
      |                 long long unsigned int
      |              %lx
plugins/scaleflux/sfx-nvme.c: In function ‘show_cap_info’:
plugins/scaleflux/sfx-nvme.c:600:49: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  600 |  printf("logic            capacity:%5lluGB(0x%llx)\n",
      |                                              ~~~^
      |                                                 |
      |                                                 long long unsigned int
      |                                              %lx
  601 |    IDEMA_CAP2GB(ctx->user_space), ctx->user_space);
      |                                   ~~~~~~~~~~~~~~~
      |                                      |
      |                                      __u64 {aka long unsigned int}
plugins/scaleflux/sfx-nvme.c:602:49: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  602 |  printf("provisioned      capacity:%5lluGB(0x%llx)\n",
      |                                              ~~~^
      |                                                 |
      |                                                 long long unsigned int
      |                                              %lx
  603 |    IDEMA_CAP2GB(ctx->phy_space), ctx->phy_space);
      |                                  ~~~~~~~~~~~~~~  
      |                                     |
      |                                     __u64 {aka long unsigned int}
plugins/scaleflux/sfx-nvme.c:604:49: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  604 |  printf("free provisioned capacity:%5lluGB(0x%llx)\n",
      |                                              ~~~^
      |                                                 |
      |                                                 long long unsigned int
      |                                              %lx
  605 |    IDEMA_CAP2GB(ctx->free_space), ctx->free_space);
      |                                   ~~~~~~~~~~~~~~~
      |                                      |
      |                                      __u64 {aka long unsigned int}
plugins/scaleflux/sfx-nvme.c:606:49: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  606 |  printf("used provisioned capacity:%5lluGB(0x%llx)\n",
      |                                              ~~~^
      |                                                 |
      |                                                 long long unsigned int
      |                                              %lx
  607 |    IDEMA_CAP2GB(ctx->phy_space) - IDEMA_CAP2GB(ctx->free_space),
  608 |    ctx->phy_space - ctx->free_space);
      |    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~              
      |                   |
      |                   __u64 {aka long unsigned int}
plugins/scaleflux/sfx-nvme.c: In function ‘change_sanity_check’:
plugins/scaleflux/sfx-nvme.c:697:5: error: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  697 |     "WARNING: Free memory is not enough! "
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
......
  700 |     mem_need, s_info.freeram);
      |     ~~~~~~~~
      |     |
      |     __u64 {aka long unsigned int}
plugins/scaleflux/sfx-nvme.c:699:35: note: format string is defined here
  699 |     "WARNING: Memory needed is %llu, free memory is %lu\n",
      |                                ~~~^
      |                                   |
      |                                   long long unsigned int
      |                                %lu
plugins/scaleflux/sfx-nvme.c: In function ‘change_cap’:
plugins/scaleflux/sfx-nvme.c:777:17: error: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  777 |  printf("%dG %lluB %llu 4K\n",
      |              ~~~^
      |                 |
      |                 long long unsigned int
      |              %lu
  778 |   cfg.capacity_in_gb, cfg.cap_in_byte, cap_in_4k);
      |                       ~~~~~~~~~~~~~~~
      |                          |
      |                          __u64 {aka long unsigned int}
plugins/scaleflux/sfx-nvme.c:777:23: error: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 4 has type ‘__u64’ {aka ‘long unsigned int’} [-Werror=format=]
  777 |  printf("%dG %lluB %llu 4K\n",
      |                    ~~~^
      |                       |
      |                       long long unsigned int
      |                    %lu
  778 |   cfg.capacity_in_gb, cfg.cap_in_byte, cap_in_4k);
      |                                        ~~~~~~~~~
      |                                        |
      |                                        __u64 {aka long unsigned int}
    CC plugins/transcend/transcend-nvme.o
    CC plugins/zns/zns.o
    CC util/argconfig.o
cc1: all warnings being treated as errors
make: *** [Makefile:95: plugins/scaleflux/sfx-nvme.o] Error 1
make: *** Waiting for unfinished jobs....
```

keithbusch, 2020-07-27:
Will get this fixed asap and add this common error check for the continuous integration.

brahads, 2020-08-25:
It fails to build on ppc64le for SuSE as well, with the same errors.

keithbusch, 2020-08-25:
Sorry, I thought I'd pushed this a while ago. Should be fixed now.

anonymouse64, 2020-08-25:
looks fixed to me, thanks!

closed 2020-08-25 by anonymouse64

keithbusch, 2020-08-25:
Thank you for confirming! And apologies for the delay; this was nearly a month old for a pretty trivial issue.