linux-nvme/nvme-cli #798: a question about the documentation manual

zhouhuaLiu, 2020-08-28:
I'd like to add a new command. I notice that in the Documentation directory, there are three styles of command manual (suffix .txt .html .1) for each command. I don't know how they were generated. I looked up the docbook_xls tool, but still confused, especially for the file with suffix ".1".
Can anyone tell me how to do and what tool I need ? Thanks very much!

zhouhuaLiu, 2020-08-28:
@keithbusch

keithbusch, 2020-08-28:
You only need to write the .txt. The rest are generated from "make doc"

keithbusch, 2020-08-28:
btw, you don't need to generate the man page. I just do that whenever a release is imminent. Normally i wouldn't include auto-generated files in source control, but there were some complaints very early from environments that didn't have asciidoc. It was just easier to include them here rather than explain how to install the build dependencies.

closed 2022-01-19 by igaw