linux-nvme/nvme-cli #800: what detail this messages means？ status_field : 0xc502(READ_ERROR

tscswcn, 2020-09-02:
................
Entry[63]
.................
error_count : 9290042
sqid : 52
cmdid : 0x221
status_field : 0xc502(READ_ERROR: The read data could not be recovered from the media)
parm_err_loc : 0
lba : 0x19408a50
nsid : 0
vs : 0
cs : 0x1010101

birkelund, 2020-09-02:
Ehm - I think it says on the tin: "Read Error" :)

birkelund, 2020-09-02:
If you are asking how that error code is encoded in`0xC502`, then its `0xC502 >> 1` to get rid of the Phase Tag. That leave us with `0x6281`. Then apply a mask of `0x7ff` to extract the lower 11 bytes (3 for the Status Code Type and 8 for the Status Code), ending up with `0x281`. `0x2xx` are "Media and Data Integrity Errors" and the `0x81` status code is "Unrecovered Read Error". But nvme-cli has already decoded this for you.

tscswcn, 2020-09-13:
thanks!  plase close this issue

closed 2020-09-13 by tscswcn