linux-nvme/nvme-cli #802: [Question] nvme fw-commit & nvme list: error 113

ThaKamus, 2020-09-03:
Hi, I have a doubt about the time response of the above commands. I tested a "home-made" script that issues the following commands in a terminal: 
`nvme fw-download /dev/nvme0n1 --fw=fw.bin --xfer=1024
nvme fw-activate /dev/nvme0n1 --slot=0 --action=3
nvme fw-download /dev/nvme1n1 --fw=fw.bin --xfer=1024
nvme fw-activate /dev/nvme1n1 --slot=0 --action=3
nvme fw-download /dev/nvme2n1 --fw=fw.bin--xfer=1024
nvme fw-activate /dev/nvme2n1 --slot=0 --action=3
nvme list`

After nvme list is issued, I get the next error:
`/dev/nvme2n1: failed to obtain nvme info: Unknown 
nvme list finished with error code 113
nvme list tool finished at 2020-09-01 22:26:17,107678 and took 0.108289 seconds to complete`

The OS is Linux-5.5.11-x86_64-with-redhat-8.1-Ootpa with Kernel: 5.5.11

It could be due terminal uses Multithread? or last command needs a delay to respond? if yes, it is expected?

Thanks,

keithbusch, 2020-09-03:
I'm not sure what to make of your output. Where `/dev/nvme2n1: failed to obtain nvme info: Unknown` getting logged?

ThaKamus, 2020-09-03:
Ohhh, @keithbusch  that was the output error from the Terminal window on RHEL.
Let me share you a screenshot about:
[image]

keithbusch, 2020-09-03:
ok, i just wasn't finding that string in the tool, so not sure what exactly it's failing to obtain.

So you're saying that `nvme list` shows all your devices without error if you add a delay? That honestly seems quite odd from the above sequence. Activating firmware shouldn't do anything to the device nodes or sysfs entries.

ThaKamus, 2020-09-03:
@keithbusch not exactly, it seems that, when you issue all the commands in a bulk, the `nvme list` **doesn't** enumerates the last drive `/dev/nvme2n1` and I'm suspecting that is because the device is in the middle of `commit action = 3`. So my question is: that is an expected behavior? or the `nvme list` needs that all the operations ends before issue the command? and if our method to issue the commands is valid; I mean, I don't think that issuing all the commands in a single line should work fine, but I didn't found on the documentation something about this. That's why I'm asking.

keithbusch, 2020-09-03:
From the tool's point of view, it should be totally fine to issue all these. What you're describing sounds like something odd with the kernel. The only thing a f/w commit should do is either stage the firmware for the next reset, or, if the device implements some of the more advanced NVMe FW activation features, it should temporarily pause IO. But nothing should fail.

Is there anything in 'dmesg' coming up during the running of these commands?

ThaKamus, 2020-09-03:
Thanks for your comments @keithbusch ; yes, my nvme devices support `commit action = 3`. I need to take a look at the 'dmesg' because with `OS:Linux-4.12.14-195-default-x86_64-with-glibc2.3 + Kernel Version:4.12.14-195-default` which is also RHEL 8.1 **the issue doesn't happen** so maybe your hint is right, the kernel is doing something odd. Later I'll update this thread with my findings.

ThaKamus, 2020-09-07:
Thanks for your help @keithbusch I confirm that the behavior I reported is a Kernel 5.5.11 issue with RHEL + NVMe module.

closed 2020-09-07 by ThaKamus

keithbusch, 2020-09-08:
> Thanks for your help @keithbusch I confirm that the behavior I reported is a Kernel 5.5.11 issue with RHEL + NVMe module.

Thank you for confirming, though that observation has me a bit nervous. Would you be able to help us on the kernel side? What I'd like to know is if this odd behavior occurs with the most recent stable kernel (5.8.7 as of today) and if so, is there anything in dmesg logged during this test?