linux-nvme/nvme-cli #805: nvme0 change to nvme1 after device powercycle

seanmaster, 2020-09-10:
Hi @keithbusch 
I have a question about linux kernel(5.8.0-1) behavior about nvme driver. 
sometimes, my test drive's device node will  be changed after device power cycle(**nvme0 -> nvme1**)
crw------- 1 root root 246, 0 Sep 10 01:50 /dev/**nvme1**
brw-rw---- 1 root disk 259, 0 Sep 10 01:50 /dev/**nvme0n1**

Below is flow:
1. IO runing
2. sudden power off (control power switch)
3. remove device (echo 1 >/sys/bus/pci/device/xxxxx/remove
4. power on(control power switch)
5. rescan device (echo 1 >/sys/bus/pci/rescan)

Not sure if this question is suitable for asking here, if not suitable, please let me know. thanks~~

keithbusch, 2020-09-10:
That means something in user space continues to hold a reference to the old nvme0 after it was removed. The controller created on the rescan has to use a new identifier, otherwise it will clash with the old one that is still being referenced.

seanmaster, 2020-09-11:
Thanks reply,
Any cmd or method to search which application keep to reference the drive nvmex?

keithbusch, 2020-09-11:
There's several ways to do that, though I don't know the best ones off the top of my head.

If you've got tracing enabled on your kernel, you could use ftrace. Setting it up prior to running your test could look like this:

```
cd /sys/kernel/debug/tracing
echo nvme_*open > set_ftrace_filter
echo nvme_*release >> set_ftrace_filter
```

After running the test, run `cat trace`. It should show the name of all processes and their PID that take and release references.

seanmaster, 2020-09-13:
Thanks @keithbusch 
I just found which process lock my test drive when power off the power.
When the device node was changed, "lsof | grep nvme0" can found "smartd" daemon was reference "nvme0" so when next power on the device node change to nvme1,  I think i can try disable the smartd first before test.
Thank you recommend about trace the kernel event.

keithbusch, 2020-09-13:
Ah, lsof works if the process by is still running. I for some reason assumed it was a short lived process, but your method works great for your observation.

seanmaster, 2020-09-15:
Thanks your help and suggest.

closed 2020-09-15 by seanmaster