linux-nvme/nvme-cli #806: why nvme intel lat-stats data didn't match perf's data

swf504, 2020-09-14:
perf's data like:
Summary latency data for PCIE (0000:00:09.0)                         from core 0:
=================================================================================
  1.00000% :     6.428us
 10.00000% :     6.511us
 25.00000% :     6.539us
 50.00000% :     6.623us
 75.00000% :     6.817us
 90.00000% :     7.068us
 95.00000% :     7.179us
 98.00000% :     7.903us
 99.00000% :    10.407us
 99.50000% :    15.249us
 99.90000% :    31.833us
 99.99000% :    36.731us
 99.99900% :    45.858us
 99.99990% :    72.571us
 99.99999% :   133.566us

And lat-stats's data like 
[root@HN_0_0 ~]# nvme intel lat-stats /dev/nvme0n1 -w
 Intel IO Write Command Latency Statistics
-------------------------------------
Major Revision : 4
Minor Revision : 1

Group 1: Range is 0-1ms, step is 32us
Bucket  0: 0                   #0~32us  
Bucket  1: 0                   #32~64us
Bucket  2: 27926401
Bucket  3: 70933315
Bucket  4: 129010370
Bucket  5: 260592977
Bucket  6: 253262697
Bucket  7: 214953419
Bucket  8: 157248744
Bucket  9: 125839540
Bucket 10: 97967792
Bucket 11: 65886311

Is the log page of 0xC1, 0xC2  incorrect?

keithbusch, 2020-09-14:
I don't understand your output at all. That's not what the code says it prints. I would have expected multiple columns of output for a version 4.1 intel lat stat.

Anyway, the lat stats logged by the controller will never match what you measure form the host. The controller's log measures media access latency. The host measures transport, command queueing and completion latency in addition to the media latency.

swf504, 2020-09-16:
@keithbusch Thanks  for your comments. 
I understand the different betweent contoller's latency and host latency.
But In my environment, I can't understand why media access latency measures by controller greater than host's laytency， The output of lat-stats show that most of latency was 128us to 288us, But the perf data which generated by SPDK  perf tool show that most of write command latency was less than 32us。

closed 2022-01-19 by igaw