linux-nvme/nvme-cli #814: NVME wear level

Badb0yBadb0y, 2020-10-01:
Hello,

I can't get the wear level of my NVME devices :( 

In the ILO module I can see:
[image]

I've taken the latest nvme-cli from git:
nvme version 1.12.112.g3aed

```
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     KI99N0461I0109U0B    VK003840KWWFP                            1           3.84  TB /   3.84  TB    512   B +  0 B   HPK1
/dev/nvme3n1     KI99N0461I0109U0G    VK003840KWWFP                            1           3.84  TB /   3.84  TB    512   B +  0 B   HPK1
/dev/nvme2n1     KI99N0461I0109U0H    VK003840KWWFP                            1           3.84  TB /   3.84  TB    512   B +  0 B   HPK1
/dev/nvme1n1     KI9AN0136I02BBI1V    VK003840KWWFP                            1           3.84  TB /   3.84  TB    512   B +  0 B   HPK1
/dev/nvme7n1     PHLN003400R81P6AGN   MO001600KWVUU                            1           1.60  TB /   1.60  TB    512   B +  0 B   4ICRHPK0
/dev/nvme6n1     PHLN003400EA1P6AGN   MO001600KWVUU                            1           1.60  TB /   1.60  TB    512   B +  0 B   4ICRHPK0
/dev/nvme4n1     KI99N0461I0109U0V    VK003840KWWFP                            1           3.84  TB /   3.84  TB    512   B +  0 B   HPK1
/dev/nvme5n1     KI9AN0136I02BBI1S    VK003840KWWFP                            1           3.84  TB /   3.84  TB    512   B +  0 B   HPK1

```

The smart-log gives me back this output:
```
Smart Log for NVME device:nvme0n1 namespace-id:ffffffff
critical_warning                        : 0
temperature                             : 29 C
available_spare                         : 100%
available_spare_threshold               : 10%
percentage_used                         : 0%
endurance group critical warning summary: 0
data_units_read                         : 8,530,403
data_units_written                      : 2,412,707
host_read_commands                      : 99,003,554
host_write_commands                     : 68,187,569
controller_busy_time                    : 112
power_cycles                            : 40
power_on_hours                          : 2,659
unsafe_shutdowns                        : 22
media_errors                            : 0
num_err_log_entries                     : 83
Warning Temperature Time                : 0
Critical Composite Temperature Time     : 0
Temperature Sensor 2           : 37 C
Temperature Sensor 4           : 55 C
Thermal Management T1 Trans Count       : 0
Thermal Management T2 Trans Count       : 0
Thermal Management T1 Total Time        : 0
Thermal Management T2 Total Time        : 0

```

Plugin for HP there isn't any in the man page, but I've tried an intel:
```
nvme intel smart-log-add /dev/nvme0n1
NVMe Status:INVALID_LOG_PAGE: The log page indicated is invalid. This error condition is also returned if a reserved log page is requested(2109)
```

If I do with lnvm it gives me back an error:
```
nvme lnvm smart-log-add /dev/nvme0n1
ERROR: Invalid sub-command 'smart-log-add' for plugin lnvm
nvme-1.12.112.g3aed
usage: nvme <command> [<device>] [<args>]

The '<device>' may be either an NVMe character device (ex: /dev/nvme0) or an
nvme block device (ex: /dev/nvme0n1).

```

Is my command wrong how I try to get the information? Or you just don't support it at the moment?

keithbusch, 2020-10-01:
What is "the information" you're specifically trying to get?

Badb0yBadb0y, 2020-10-01:
@keithbusch 

Some information about how many % left from the lifetime of the nvme or even better how many days left before it dies like on the screenshot above.

keithbusch, 2020-10-01:
The smart logs available spare?

keithbusch, 2020-10-01:
Between percentage used and available spare, I think you have everything you're asking for. Am I missing something? That's pretty much all that's provided by spec.

keithbusch, 2020-10-01:
It's just an indication of "wear". How "level" that wear is depends on the device's FTL implementation is.

closed 2020-10-01 by keithbusch

Badb0yBadb0y, 2020-10-02:
Yeye, but some indicator that I want to use is this actually.

Thank you the clarification.

Badb0yBadb0y, 2020-10-05:
Hi,

can reopen the issue please?
https://www.percona.com/blog/2017/02/09/using-nvme-command-line-tools-to-check-nvme-flash-health/

So that is not the one that I'm looking for. You can see on this website it is regarding space, but I'm looking for wear level which means how many days or percentage left from the nvme life.

keithbusch, 2020-10-05:
You're asking about a non-standard log. The vendor needs to provide support for that.

Badb0yBadb0y, 2020-10-06:
I've just checked the nvme drives, some of them intel based, some of the hynix based.
Intel can show some information like this:

```
nvme intel smart-log-add /dev/nvme6n1
Additional Smart Log for NVME device:nvme6n1 namespace-id:ffffffff
ID             KEY                                 Normalized     Raw
0xab    program_fail_count                             100         0
0xac    erase_fail_count                               100         0
0xad    wear_leveling_count                            100         4295098369
0xb8    e2e_error_detect_count                         100         0
0xc7    crc_error_count                                100         0
0xe2    media_wear_percentage                          100         65535
0xe3    host_reads                                     100         65535
0xe4    timed_work_load                                100         65535
0xea    thermal_throttle_status                        100         0
0xf0    retry_buff_overflow_count                      100         0
0xf3    pll_lock_loss_counter                          100         0
0xf4    nand_bytes_written                             100         40262
0xf5    host_bytes_written                             100         45197
0xf6    host_context_wear_used                         100         0
```

However do you have support for hynix?

keithbusch, 2020-10-06:
You need to ask hynix that. You're asking about a non standard log page, support for vendor specific logs has to come from the vendor. There's nothing this tool can do about it without vendor support.