linux-nvme/nvme-cli #816: nvme format failing for Hynix

way2ezee, 2020-10-03:
Please help me to troubleshoot this problem.

Command Output:-
```
ubuntu@ubuntu:/$ sudo nvme format /dev/nvme0 --lbaf=1
Invalid namespace ID, specify a namespace to format or use '-n 0xffffffff' to format all namespaces on this controller.

ubuntu@ubuntu:/$ sudo nvme format /dev/nvme0 --lbaf=1 -n 1
NVMe status: INVALID_FORMAT: The LBA Format specified is not supported. This may be due to various conditions(0x10a)

```
```
ubuntu@ubuntu:/$ sudo nvme id-ns /dev/nvme0n1
NVME Identify Namespace 1:
nsze    : 0x3b9e12b0
ncap    : 0x3b9e12b0
nuse    : 0x3b9e12b0
nsfeat  : 0
nlbaf   : 1
flbas   : 0
mc      : 0
dpc     : 0
dps     : 0
nmic    : 0
rescap  : 0
fpi     : 0
dlfeat  : 0
nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 0
nsattr	: 0
nvmsetid: 0
anagrpid: 0
endgid  : 0
nguid   : ace42e000a13c0112ee4ac0000000001
eui64   : ace42e000a13c011
lbaf  0 : ms:0   lbads:9  rp:0 (in use)
lbaf  1 : ms:0   lbads:12 rp:0 
```
```
ubuntu@ubuntu:~$ sudo smartctl -a /dev/nvme0
smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.4.0-42-generic] (local build)
Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       SKHynix_HFS512GD9TNG-L5B0B
Serial Number:                      XXXXXXXXXXXX
Firmware Version:                   80750C10
PCI Vendor/Subsystem ID:            0x1c5c
IEEE OUI Identifier:                0xace42e
Controller ID:                      1
Number of Namespaces:               1
Namespace 1 Size/Capacity:          512,110,190,592 [512 GB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            ace42e 000a13c011
Local Time is:                      Sat Oct  3 11:14:57 2020 UTC
Firmware Updates (0x16):            3 Slots, no Reset required
Optional Admin Commands (0x0017):   Security Format Frmw_DL Self_Test
Optional NVM Commands (0x005e):     Wr_Unc DS_Mngmt Wr_Zero Sav/Sel_Feat Timestmp
Maximum Data Transfer Size:         64 Pages
Warning  Comp. Temp. Threshold:     86 Celsius
Critical Comp. Temp. Threshold:     87 Celsius

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +   6.3000W       -        -    0  0  0  0        5       5
 1 +   2.4000W       -        -    1  1  1  1       30      30
 2 +   1.9000W       -        -    2  2  2  2      100     100
 3 -   0.0500W       -        -    3  3  3  3     1000    1000
 4 -   0.0040W       -        -    3  3  3  3     1000    9000

Supported LBA Sizes (NSID 0x1)
Id Fmt  Data  Metadt  Rel_Perf
 0 +     512       0         0
 1 -    4096       0         0

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02)
Critical Warning:                   0x00
Temperature:                        49 Celsius
Available Spare:                    100%
Available Spare Threshold:          10%
Percentage Used:                    0%
Data Units Read:                    2,842,547 [1.45 TB]
Data Units Written:                 1,701,565 [871 GB]
Host Read Commands:                 11,828,131
Host Write Commands:                6,840,613
Controller Busy Time:               124
Power Cycles:                       221
Power On Hours:                     70
Unsafe Shutdowns:                   155
Media and Data Integrity Errors:    0
Error Information Log Entries:      0
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
Temperature Sensor 1:               49 Celsius
Temperature Sensor 2:               52 Celsius

Error Information (NVMe Log 0x01, max 256 entries)
No Errors Logged

```
Thanks

keithbusch, 2020-10-03:
Do you only have one nvme device? If not, make sure nvme0 really is the parent controller to nvme0n1. Sometimes it's not. So if you have more than one nvme, try to use the nvme0n1 handle consistently rather than sometimes using nvme0.

If you have only one, or are sure nvme0 is the parent to nvme0n1, then the 2nd format command looks fine. Does it work if you use lbaf 0?

If not, try using namespace id 0xffffffff. Some controllers just don't allow formatting a specific namespace, even if they only have one. The 'id-ctrl' fna field should indicate if this is one of those types, though that's not the right return code even if this is one of those kinds of controllers.

way2ezee, 2020-10-03:
I have only 1 drive in this laptop.
With 0xffffffff, same output
```
ubuntu@ubuntu:~$ sudo nvme format /dev/nvme0 -n 0xffffffff -l 1
NVMe status: INVALID_FORMAT: The LBA Format specified is not supported. This may be due to various conditions(0x10a)
```
```
ubuntu@ubuntu:~$ sudo nvme id-ctrl /dev/nvme0
NVME Identify Controller:
vid       : 0x1c5c
ssvid     : 0x1c5c
sn        : ND02N6627102Y2P31   
mn        : SKHynix_HFS512GD9TNG-L5B0B              
fr        : 80750C10
rab       : 4
ieee      : ace42e
cmic      : 0
mdts      : 6
cntlid    : 0x1
ver       : 0x10300
rtd3r     : 0x7a120
rtd3e     : 0x1e8480
oaes      : 0x200
ctratt    : 0
rrls      : 0
crdt1     : 0
crdt2     : 0
crdt3     : 0
oacs      : 0x17
acl       : 3
aerl      : 7
frmw      : 0x16
lpa       : 0x2
elpe      : 255
npss      : 4
avscc     : 0x1
apsta     : 0x1
wctemp    : 359
cctemp    : 360
mtfa      : 0
hmpre     : 0
hmmin     : 0
tnvmcap   : 0
unvmcap   : 0
rpmbs     : 0
edstt     : 60
dsto      : 1
fwug      : 0
kas       : 0
hctma     : 0x1
mntmt     : 273
mxtmt     : 358
sanicap   : 0x3
hmminds   : 0
hmmaxd    : 0
nsetidmax : 0
anatt     : 0
anacap    : 0
anagrpmax : 0
nanagrpid : 0
sqes      : 0x66
cqes      : 0x44
maxcmd    : 0
nn        : 1
oncs      : 0x5e
fuses     : 0
fna       : 0x6
vwc       : 0x1
awun      : 0
awupf     : 0
nvscc     : 1
nwpc      : 0
acwu      : 0
sgls      : 0
mnan      : 0
subnqn    : 
ioccsz    : 0
iorcsz    : 0
icdoff    : 0
ctrattr   : 0
msdbd     : 0
ps    0 : mp:6.3000W operational enlat:5 exlat:5 rrt:0 rrl:0
          rwt:0 rwl:0 idle_power:- active_power:-
ps    1 : mp:2.4000W operational enlat:30 exlat:30 rrt:1 rrl:1
          rwt:1 rwl:1 idle_power:- active_power:-
ps    2 : mp:1.9000W operational enlat:100 exlat:100 rrt:2 rrl:2
          rwt:2 rwl:2 idle_power:- active_power:-
ps    3 : mp:0.0500W non-operational enlat:1000 exlat:1000 rrt:3 rrl:3
          rwt:3 rwl:3 idle_power:- active_power:-
ps    4 : mp:0.0040W non-operational enlat:1000 exlat:9000 rrt:3 rrl:3
          rwt:3 rwl:3 idle_power:- active_power:-
```

way2ezee, 2020-10-03:
I have only 1 drive in this laptop.
With 0xffffffff, same output
```
ubuntu@ubuntu:~$ sudo nvme format /dev/nvme0 -n 0xffffffff -l 1
NVMe status: INVALID_FORMAT: The LBA Format specified is not supported. This may be due to various conditions(0x10a)
```
```
ubuntu@ubuntu:~$ sudo nvme id-ctrl /dev/nvme0
NVME Identify Controller:
vid       : 0x1c5c
ssvid     : 0x1c5c
sn        : NDXXXXXXXXXXXXXXXXXX
mn        : SKHynix_HFS512GD9TNG-L5B0B              
fr        : 80750C10
rab       : 4
ieee      : ace42e
cmic      : 0
mdts      : 6
cntlid    : 0x1
ver       : 0x10300
rtd3r     : 0x7a120
rtd3e     : 0x1e8480
oaes      : 0x200
ctratt    : 0
rrls      : 0
crdt1     : 0
crdt2     : 0
crdt3     : 0
oacs      : 0x17
acl       : 3
aerl      : 7
frmw      : 0x16
lpa       : 0x2
elpe      : 255
npss      : 4
avscc     : 0x1
apsta     : 0x1
wctemp    : 359
cctemp    : 360
mtfa      : 0
hmpre     : 0
hmmin     : 0
tnvmcap   : 0
unvmcap   : 0
rpmbs     : 0
edstt     : 60
dsto      : 1
fwug      : 0
kas       : 0
hctma     : 0x1
mntmt     : 273
mxtmt     : 358
sanicap   : 0x3
hmminds   : 0
hmmaxd    : 0
nsetidmax : 0
anatt     : 0
anacap    : 0
anagrpmax : 0
nanagrpid : 0
sqes      : 0x66
cqes      : 0x44
maxcmd    : 0
nn        : 1
oncs      : 0x5e
fuses     : 0
fna       : 0x6
vwc       : 0x1
awun      : 0
awupf     : 0
nvscc     : 1
nwpc      : 0
acwu      : 0
sgls      : 0
mnan      : 0
subnqn    : 
ioccsz    : 0
iorcsz    : 0
icdoff    : 0
ctrattr   : 0
msdbd     : 0
ps    0 : mp:6.3000W operational enlat:5 exlat:5 rrt:0 rrl:0
          rwt:0 rwl:0 idle_power:- active_power:-
ps    1 : mp:2.4000W operational enlat:30 exlat:30 rrt:1 rrl:1
          rwt:1 rwl:1 idle_power:- active_power:-
ps    2 : mp:1.9000W operational enlat:100 exlat:100 rrt:2 rrl:2
          rwt:2 rwl:2 idle_power:- active_power:-
ps    3 : mp:0.0500W non-operational enlat:1000 exlat:1000 rrt:3 rrl:3
          rwt:3 rwl:3 idle_power:- active_power:-
ps    4 : mp:0.0040W non-operational enlat:1000 exlat:9000 rrt:3 rrl:3
          rwt:3 rwl:3 idle_power:- active_power:-
```

way2ezee, 2020-10-03:
f```
na       : 0x6
  [2:2] : 0x1	Crypto Erase Supported as part of Secure Erase
  [1:1] : 0x1	Crypto Erase Applies to All Namespace(s)
  [0:0] : 0	Format Applies to Single Namespace(s)
```

keithbusch, 2020-10-03:
I think you'd have to open a bug with your vendor at this point.

way2ezee, 2020-10-03:
Any other way to change sector size?

keithbusch, 2020-10-03:
No other spec defined way. Your drive is simply not behaving to the spec, so I'd guess it's a drive firmware bug. You'll need to contact the vendor for a solution.

tim-oleksii, 2020-10-28:
@way2ezee is it pc401 drive?

keithbusch, 2020-10-28:
> @way2ezee is it pc401 drive?

It may appear so based on the model number, though I am not sure what the L5B0B suffix indicates. That actually looks closer aligned to the PC601 from what I can gather from the public listings on skhynix.com.

tim-oleksii, 2020-10-28:
Seems like pc601 judging by sn. I'd try to contact Lenovo & hynix simultaneously.

way2ezee, 2020-10-28:
pc601

tim-oleksii, 2020-10-28:
@way2ezee Sorry, my comment may look misleading - please e-mail them. As it's PC601, it might get fixed.

way2me, 2020-11-02:
They do not take customers seriously
https://forums.lenovo.com/t5/ThinkPad-T400-T500-and-newer-T-series-Laptops/nvme-format-failing-with-4096-sector-size/m-p/5043069?page=1#5160285

keithbusch, 2020-11-02:
Don't take it personally. Your experience so far is par for the course with any vendor.

keithbusch, 2020-11-02:
BTW, they seem to think it could be a security setting on the device. That could certainly prevent a format command, but I don't think they're using the correct error code. Security sequences should use the "Command Sequence Error".

On a side note, while the 4k format provides all the right guarantees for more optimal access granularity, as long as you partition your device on a 4k boundary, most Linux filesystems will use 4k aligned access too.

TheLQ, 2021-12-03:
Thank you @tomty89 can confirm that silly workaround worked

```
# nvme id-ns /dev/nvme0 --namespace-id=1 -H
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0 Best (in use)
LBA Format  1 : Metadata Size: 0   bytes - Data Size: 4096 bytes - Relative Performance: 0 Best

# nvme format /dev/nvme0n1 --lbaf=1
You are about to format nvme0n1, namespace 0x1.
Namespace nvme0n1 has parent controller(s):nvme0
...
Sending format operation ...
NVMe status: INVALID_FORMAT: The LBA Format specified is not supported. This may be due to various conditions(0x10a)

# systemctl suspend
...physically wake machine....

# nvme format /dev/nvme0 --lbaf=1
You are about to format nvme0, namespace 0xffffffff(ALL namespaces).
Controller nvme0 has child namespace(s):nvme0n1
Sending format operation ...
Success formatting namespace:ffffffff
failed to re-read partition table
```

My setup in case it helps

* Asus z170a motherboard. Running without secure boot
* Retail sold as Hynix P31 like OP

igaw, 2022-01-19:
Doesn't seem there is anything we can do here. Closing the issue, if you think is still relevant please reopen.

closed 2022-01-19 by igaw

ku1ik, 2023-10-08:
Had the same problem with WD SN700, the suspend trick worked for me :+1:

aggieNick02, 2024-06-19:
The suspend/resume trick has worked for me as well on a number of motherboards, but not all. So YMMV, depending on the motherboard.

aggieNick02, 2024-06-21:
Please see my comment on this related issue [here](https://github.com/linux-nvme/nvme-cli/issues/84#issuecomment-2183511051). It covers what is happening, why the suspend/resume trick often works, and what you can do when it does not.