linux-nvme/nvme-cli #828: nvme-dsm command with a single range >2TB returns successful status code

alexeyponomarev, 2020-10-30:
Currently it's possible to send a dsm command with nvme-cli specifying a single range (--blocks) of more than 2TB. Since the range is limited by 32 bits in the tool/spec, nvme-dsm currently silently ignores the portion of the value that is >32 bits returning successful status code/no warning for the user.

For example, a command to trim 7501476528 blocks (3,576GB) will return successful status code:

# ./nvme dsm /dev/nvme0n1 --blocks 7501476528 --slbs 0 -a 0 -d
 NVMe DSM: success

But in reality the tool will send a DSM command for only 3,206,509,232 blocks (1,528GB) since it will ignore bit 32 set in the 7501476528 value:

(gdb) r dsm /dev/nvme0n1 --blocks 7501476528 --slbs 0 -a 0 -d
Starting program: /tmp/nvme-cli-master/nvme dsm /dev/nvme0n1 --blocks 7501476528 --slbs 0 -a 0 -d

Breakpoint 1, nvme_setup_dsm_range (ctx_attrs=0x7fffffffdbd0, llbas=0x7fffffffd7d0, slbas=0x7fffffffcfd0, nr_ranges=1) at nvme-ioctl.c:258
258             struct nvme_dsm_range *dsm = malloc(nr_ranges * sizeof(*dsm));
...
(gdb) s
nvme_setup_dsm_range (ctx_attrs=0x7fffffffdbd0, llbas=0x7fffffffd7d0, slbas=0x7fffffffcfd0, nr_ranges=1) at nvme-ioctl.c:267
267                     dsm[i].slba = cpu_to_le64(slbas[i]);
(gdb) p dsm[i].nlb
$1 = 3206509232

This is done silently without any warnings or errors for the user, who might think that since the command finished with successful status code - a full range they requested was trimmed.

keithbusch, 2020-10-30:
True, we're using strtol, which isn't going to produce an error converting that string. Let me see what I can do.

keithbusch, 2020-10-30:
Should be fixed now

closed 2020-10-30 by keithbusch