linux-nvme/nvme-cli #829: reservation command execution failed when there are multiple namespaces

ryderhsu0129, 2020-11-03:
The resv-register command is executed successfully when there is only one namespace, for example :
`nvme resv-register /dev/nvme0 -n 1 -k 1 -r 0`

but the command execution failed when there are multiple namespaces,
and error message on dmesg is shown as:
`NVME_IOCTL_IO_CMD not supported when multiple namespaces present!`
the reason seems to be that controller handle(/dev/nvme0) cannot be used when there are multiple namespace.

NVMe reservations feature claim that the shared namespace can be accessed by multiple hosts(controllers), If we cannot use controller handle(/dev/nvme0) we will not be able to specify the specific controller to access the shared namespace.
Is there any other way to achieve this?
thanks!

keithbusch, 2020-11-03:
Reservations are IO commands issued to namespaces. You should execute the command to those rather than the controller. A namespace handle looks like /dev/nvme0n1, for example.

ryderhsu0129, 2020-11-03:
@keithbusch 

thanks for your reply.

If we has two controller handle(/dev/nvme0、/dev/nvme1),
and attached a shared namespace to them,
How do we issue reservation command to the specified controller?

> Reservations are IO commands issued to namespaces. You should execute the command to those rather than the controller. A namespace handle looks like /dev/nvme0n1, for example.

keithbusch, 2020-11-03:
Reservations are for the subsystem rather than controller. You should have the same host id registered with all the controllers the host connects to. Then it doesn't matter which controller receive the reservation command.

ryderhsu0129, 2020-11-03:
OK, thanks!

closed 2020-11-03 by ryderhsu0129

ryderhsu0129, 2020-11-05:
> Reservations are for the subsystem rather than controller. You should have the same host id registered with all the controllers the host connects to. Then it doesn't matter which controller receive the reservation command.

@keithbusch 

I want to ask a similar question,
If an NVM subsystem has two controllers and a shared namespace,
can I specify the controller to do read or write to the namespace?
or can it only be determined by the linux kernel?

keithbusch, 2020-11-05:
If you're using the kernel's native nvme multipathing (which is usually enabled by default on most modern kernels), the kernel decides which path (i.e. controller) an IO command is dispatched to.

If you don't have native multipathing enabled, then you get unique handles to each path, so you get to decide which controller gets the read or write command.

ryderhsu0129, 2020-11-06:
thanks for your answer!