linux-nvme/nvme-cli #831: get-features interaction with -b option

trinof, 2020-11-04:
Maybe I'm misinterpreting how this option interacts with different commands, but issuing
`nvme get-feature /dev/nvme0 -f 7 -b`
returns 
`get-feature:0x7 (Number of Queues), Current value:0x0f000f`
exactly the same output as without the -b option.
I also tried the same thing with the generic admin-passthru command and it acts even more unexpectedly.
`nvme admin-passthru /dev/nvme0 --opcode=0x0A --cdw10=7 --namespace-id=0xFFFFFFFF -r -b`
returns nothing while
`nvme admin-passthru /dev/nvme0 --opcode=0x0A --cdw10=7 --namespace-id=0xFFFFFFFF -r`
returns
`NVMe command result:000f000f`
which is returned in stderr for whatever reason.
I would expect from both commands to receive the encoded 0x000f000f from stdout when the binary option is used.
This seems to be the behaviour in both version 1.13.11.g6d5e and 1.8.1
Thank you for your attention, and sorry for any misunderstandings I have.

keithbusch, 2020-11-04:
That's an output option if the feature uses a buffer. Very few of them do. For example, LBA Range Type is one such feature.
If they don't use a buffer, then you just get the CQE result.

The admin passthru requires you tell it how large the buffer is, otherwise you won't get any output at all when you specify output options. If you don't specify output options, the default for passthru is to just print the CQE result.

trinof, 2020-11-04:
OK, thank you for the quick response. It still seems odd that the default passthru output would go to stderr, but I think I understand the option interaction now.

closed 2020-11-04 by trinof

keithbusch, 2020-11-04:
It defaults to stderr for piping. A passthrough data in command is going to return opaque data that nvme-cli has no idea how to parse, so the result is printed to a different stream so that the data payload can be piped to a different program for parsing.