linux-nvme/nvme-cli #832: NVMF host will not auto add block device for new added namespace

the-castle, 2020-11-12:
Hi,

while the NVMF host has established a connection to one subsystem, if I create a new namespace to the subsystem on the target side, is there some way for the host to auto add block device for the new namespace? 

From my test, I can see namespace rescan is triggered on the host, but no udev rules triggered.

keithbusch, 2020-11-12:
A udev rule for the nvme block device should be triggered if the new namespaces is discovered and successfully created during the rescan. Can you verify if that part was successful?

the-castle, 2020-11-13:
Thanks Keith for your reply. I found this issue only exists when ANA is enabled. Without ANA, the host can see the associated block device.

I checked the host driver, and found the cause in nvme_mpath_add_disk : if ANA is enabled, when handling the namespace change, host driver will check the ANA log before adding the disk to the subsystem. But as the target didn't notify any ANA change event, the host would not try to get the ANA log, so the disk not added to the subsystem.

If I update the ANA state on the target side, then the host will add the disk to the subsystem, and I can see the block device.

I think there is a gap between the host driver and the target regarding on the ANA implementation. And maybe we can fix it in two ways,
1. the host driver should initiatively get the ANA log when ANA is enabled;
2. the target should notify an ANA log change event after the namespace change event.

I think the second one should be better, as the ANA state is always set by the target side and the target owns the duty to notify the change, even though this is the initial setting when the namespace is created.

What I am testing is SPDK nvmf target, I will check Linux target later.

Hope to know your views. Thanks!

jiewei-ke, 2020-11-13:
SPDK community is agreed to fix this issue on the target side. Feel free to close this issue.

closed 2020-11-13 by keithbusch

keithbusch, 2020-11-13:
> I think there is a gap between the host driver and the target regarding on the ANA implementation. And maybe we can fix it in two ways,
> 
> 1. the host driver should initiatively get the ANA log when ANA is enabled;

We do: see `nvme_mpath_init()`.

keithbusch, 2020-11-13:
> > I think there is a gap between the host driver and the target regarding on the ANA implementation. And maybe we can fix it in two ways,
> > 
> > 1. the host driver should initiatively get the ANA log when ANA is enabled;
> 
> We do: see `nvme_mpath_init()`.

And BTW, if you are observing a host-side problem initializing the ANA log for nvme's native mulitpathing, it would be better to direct the discussion to the linux nvme kernel mailing list at linux-nvme@lists.infradead.org.