linux-nvme/nvme-cli #834: How to get total capacity of /dev/nvmeN device or controller?

nxdiamond, 2020-11-17:
nvme id-ns retrieves nsze and ncap but nvme id-ctrl doesn't retrieve the total capacity of the device.  How can I find the maximum capacity to set when creating a new namespace?

keithbusch, 2020-11-17:
The total unallocated capacity of the subsystem that can potentially be used for a newly created namespace is available in `id-ctrl` unvmcap field.

nxdiamond, 2020-11-17:
unvmcap : 0
I understand now, that is the UNALLOCATED capacity but there is no report of total capacity.  Either I can delete all existing namespaces first and retrieve the total capacity, or first add unvmcap plus the ncap attributes of all existing namespaces.  For my purposes I think I need to do the latter.

nxdiamond, 2020-11-17:
Now I found this:
https://nvmexpress.org/wp-content/uploads/NVM-Express-1_2a.pdf

Both unvmcap and tnvmcap are supposed to be supported under identical conditions, "if Namespace
Management and Namespace Attachment commands are supported."

So when unvmcap is 0 (a valid output) and tnvmcap is 0 (an invalid output), is something wrong with the device?

keithbusch, 2020-11-17:
That means your device doesn't support namespace management.

keithbusch, 2020-11-17:
So I was trying to answer your second question, "the maximum capacity to set when creating a new namespace", which is reported from unvmcap. But since your device appears to not support namespace management (you can verify that with bit 3 from `id-ctrl` oacs field, or use -H to expand all the fields and look for the "NS Management and Attachment" line for support reporting), unvmcap will be 0.

But to the subject's question on the total capacity of your device, you just have to add up all the namespaces' reported capacities.

nxdiamond, 2020-11-17:
OK, if NS Management and Attachment is 0 then I should leave the namespaces alone (there probably only is one), but if NS Management and Attachment is 1 then I can delete the existing namespaces and use tnvmcap to choose the capacity for the one new namespace that I will create.

Now I think it's solved.  Thank you.

closed 2020-11-17 by nxdiamond