linux-nvme/nvme-cli #835: How to test async event request command correctly?

ryderhsu0129, 2020-11-18:
I set up a situation try to test AER command, 
and expected a reservation notification event to be triggered but did not happen, the proceed as follows:
(two controllers with one shared namespace in subsystem, each controller has their host ID)

1. register reservation keys for each controller
```
    nvme resv-register /dev/nvme0 -n 1 -k 33 -r 0
    nvme resv-register /dev/nvme1 -n 1 -k 1 -r 0
```
2. acquire a reservation with reservation type 3 for /dev/nvme1
```
nvme resv-acquire /dev/nvme1 -n 1 -c 1 -t 3 -a 0
```
3. issue a AER command to monitor reservation release event by /dev/nvme0
```
nvme admin-passthru /dev/nvme0 --opcode 0x0c
```
4. /dev/nvme1 preempt itself changing reservation type to 2
(It should produce a reservation release event)
```
nvme resv-acquire /dev/nvme1 -n 1 -p 1 -c 1 -t 2 -a 1
```

After the above steps are completed, I only receive a message :
`NVMe Status : Unknown(371)`
It looks like the kernel has timed out?
Please help clarify this issue, thanks!

keithbusch, 2020-11-18:
Aer handling is owned by the kernel. Sending it from userspace will either timeout or cause the kernel to miss an event where it needs to take action if the controller decided to respond to your command instead of the kernel's. That's why there isn't a convenient command line function for aer in this program.

For async events the kernel doesn't handle, it should becreating a uevent for the controller object and a message in the syslog.

ryderhsu0129, 2020-11-18:
@keithbusch 

1. you mean because AER handling is owned by the kernel,
so sending AER command from userspace will NOT receive respond of controller?(e.g. nvme-cli)

2. If the above is true, how can I verify the controller's AER implementation is work?

thanks!

keithbusch, 2020-11-18:
I provided you two ways the kernel notifies user space of aer events.

closed 2020-11-19 by ryderhsu0129

palapalaqoo, 2022-07-14:
If you trigger 2 times, first AER will be handled by the kernel and second will be received in your AER command.
But that is not convenient for testing AER command and kernel only print something like
'nvme_async_event: nvme0: NVME_AEN=0x000007 [NVME_AER_VS]',
If kernel print more information such as complete q and status, that will be helpfull.

igaw, 2022-07-14:
What about ftrace (events/nvme/nvme_async_event)?

palapalaqoo, 2022-07-14:
'nvme_async_event: nvme0: NVME_AEN=0x000007 [NVME_AER_VS]'  is the result from ftrace (events/nvme/nvme_async_event)
It only have event types but no completion queue or status

igaw, 2022-07-14:
Ah, I see. Anyway, you have to ask the upstream kernel mailing list about this. I suppose extending the ftrace event is more likely to be accepted than adding a kernel log entry.