linux-nvme/nvme-cli #837: Western Digital PC SN520 NVMe SSD firmware: 20370001.fluf fails to flash

Ntemis, 2020-11-27:
mint@mint:~/Downloads/FWNVA40w/WDC$ sudo nvme fw-download /dev/nvme0 --fw=20370001.fluf 
Firmware download success
mint@mint:~/Downloads/FWNVA40w/WDC$ sudo nvme fw-commit /dev/nvme0 --slot=0 --action=2
NVMe status: CMD_SEQ_ERROR: The command was aborted due to a protocol violation in a multicommand sequence(0x400c)
mint@mint:~/Downloads/FWNVA40w/WDC$ sudo nvme fw-commit /dev/nvme0 --slot=0 --action=1
NVMe status: CMD_SEQ_ERROR: The command was aborted due to a protocol violation in a multicommand sequence(0x400c)

Ntemis, 2020-12-01:
Any way to force the update please?

keithbusch, 2020-12-01:
That error code usually means there's a security block on the device. Those are vendor implementation specific, so I think you need to go to the vendor for more info, or start with the source from where you got the firmware binary.