linux-nvme/nvme-cli #844: Data corruption when using multiple devices with nvme-over-tcp

pkuwangh, 2020-12-19:
On target side, I exported 2 nvme devices using tcp/ipv6: 
```
[root@rtptest34337.prn2 ~/ext_nvme]# ll /sys/kernel/config/nvmet/ports/1/subsystems/
total 0
lrwxrwxrwx 1 root root 0 Dec 19 02:08 nvmet-rtptest34337-1 -> ../../../../nvmet/subsystems/nvmet-rtptest34337-1
lrwxrwxrwx 1 root root 0 Dec 19 02:08 nvmet-rtptest34337-2 -> ../../../../nvmet/subsystems/nvmet-rtptest34337-2_
```

On initiator side, I could successfully connect the 2 nvme devices: 
```
[root@rtptest34206.prn2 /]# nvme list
Node             SN                   Model                        Namespace          Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     ***********     INTEL *******          1         256.06  GB / 256.06  GB    512   B +  0 B    PSF119D
/dev/nvme1n1     ***********     Linux                        1         900.19  GB / 900.19  GB      4 KiB +  0 B     5.2.9-0_
/dev/nvme2n1     ***********     Linux                       1         900.19  GB / 900.19  GB      4 KiB +  0 B     5.2.9-0_
```

Then for each of nvme1n1 & nvme2n1, I created a partition; type is "linux raid autodetect"; 
Next I created a RAID-0 volume, created a filesystem on it, and mounted it. 
`mdadm --create /dev/md5 --level=0 --raid-devices=2 --chunk=128 /dev/nvme1n1p1 /dev/nvme2n1p1`
`mkfs.xfs -f /dev/md5`
`mkdir /flash; mount -o rw,noatime,discard /dev/md5 /flash/`

Now, when I copy a large directory into `/flash/`, a lot of files under /flash/ are corrupted. Specifically, that large directory has a lot of .gz files, and unzip will fail on many of them; also diff with original files does show they are different, although the file size is exactly the same. 

I'm wondering if there is a known issue, or I'm doing something not really supported. 
Thanks!

keithbusch, 2020-12-19:
This sort of sighting would be better directed to the kernel mailing list, linux-nvme@lists.infradead.org

pkuwangh, 2020-12-22:
@keithbusch, makes sense. will post there and closing this one.

closed 2020-12-22 by pkuwangh