linux-nvme/nvme-cli #846: How to test Virtualization Management of the drive

ryderhsu0129, 2020-12-23:
I know that the current tool doesn't directly support  Virtualization Management command,

If I issue this command through admin passthru, Is this a feasible way to verify the command?

or these virtual resources are handled and allocated by the kernel, commands issued by userspace will cause confusion,

just like I/O queue resources shouldn’t be created by userspace command.

keithbusch, 2020-12-23:
Please run `nvme virt-mgmt --help` for usage details.

ryderhsu0129, 2020-12-24:
@keithbusch

Thanks for your assistance.

If I run this command to change the allocation of virtual resources, like virtual queue,
does the kernel know how to handle with these queue?thanks!

JunKang, 2020-12-24:
@ryderhsu0129 
`nvme virt-mgmt`command has no effect on kernel. Its simply informs to the NVMe device.
When writing the number of virtual functions to `sriov_numvfs` file, kernel try to initiate each virtual functions and at that time reads the resource value from device that you configured before.

ryderhsu0129, 2020-12-24:
> @ryderhsu0129
> `nvme virt-mgmt`command has no effect on kernel. Its simply informs to the NVMe device.
> When writing the number of virtual functions to `sriov_numvfs` file, kernel try to initiate each virtual functions and at that time reads the resource value from device that you configured before.

@JunKang 

thanks for your reply.

If I enable SR-IOV through writing the number of virtual functions to sriov_numvfs FIRST,
then I switch the VFs to offline state for re-allocating virtual resources(issue **nvme vir-mgmt**),
Will the kernel update these configurations after switch VFs to online state without re-enable SR-IOV?

keithbusch, 2020-12-24:
Virtual resources can only be assigned to the secondary (virtual function) controllers when they're offline, so the kernel doesn't know about those controllers while you're setting their queue and interrupt resources. The kernel will react accordingly when those controllers are brought online.

ryderhsu0129, 2020-12-24:
> Virtual resources can only be assigned to the secondary (virtual function) controllers when they're offline, so the kernel doesn't know about those controllers while you're setting their queue and interrupt resources. The kernel will react accordingly when those controllers are brought online.

@keithbusch 
thanks a lot!

closed 2020-12-24 by ryderhsu0129

nguoido, 2023-04-25:
Hi @ryderhsu0129  Please tell me how to know test SR-IOV on  Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983
Thanks