linux-nvme/nvme-cli #849: From CLI commands from client(nvme-of) as well as on host, SSD usage is showing full even it is not full

smitkothari, 2021-01-05:
Hi,
I am using 250GB samsung 960 evo SSD on which I'm facing following issues/confusions.

1. When we are trying to list down the available SSDs from Client(Initiator) system using CLI command **nvme-list**, the above mentioned SSD usage field seems to be full(i.e, 250/250GB) even thought it is **empty**. Sometimes, it is the same case when tried on Host system as well.
 
2. The same CLI command when issued from Client system, the field like **Model** & **serial number** seems to be different from what we obtain on the host system.
  
Following tables are output on both host and client systems when issued nvme list command respectively.

**Host output1:**
Node                        SN                   Model                                    Namespace                    Usage                      Format                      FW Rev   
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------
/dev/nvme0n1     S3ESNX0J917822J      Samsung SSD 960 EVO 250GB                1        250.06  GB / 250.06  GB    512   B +  0 B   2B7QCXE7

**Client output1**
Node                                 SN                   Model                            Namespace   Usage                      Format           FW Rev   
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- 
/dev/nvme1n1     19e5306ae46aafd7     Linux                                    10        250.06  GB / 250.06  GB    512   B +  0 B   5.4.0-47

**Host output2:**
Node                                   SN                          Model                              Namespace        Usage                              Format           FW Rev   
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- 
/dev/nvme0n1     S3ESNX0J917822J      Samsung SSD 960 EVO 250GB                1           0.00   B / 250.06  GB    512   B +  0 B   2B7QCXE7

**Client output2**
Node                                 SN                   Model                            Namespace   Usage                      Format           FW Rev   
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- 
/dev/nvme1n1     19e5306ae46aafd7     Linux                                    10        250.06  GB / 250.06  GB    512   B +  0 B   5.4.0-47

keithbusch, 2021-01-05:
The usage refers to the NVMe Identify Controller "NUSE" field. It refers to how much of the capacity is provisioned, and most SSDs are going to say 100%.

smitkothari, 2021-01-06:
According to NVMe specs "NUSE" in identify-ns refers to total number blocks currently allocated (meaning used), "NCAP" will give the available logical blocks that can be used from the total size.(provisioned)

Also when connected over fabrics why can i not see model name as "Samsung" ?

keithbusch, 2021-01-06:
Allocated does not mean used.

If you don't have a thin provisioning controller, the controller is allowed to report NUSE as the same as NCAP at all times, which will usually be the same as NSZE.

The identify command returns the model of the target. If you're connecting to a fabrics target, the identify is going to return the model of that target rather than the backing storage.

smitkothari, 2021-01-07:
Hi Keith,
Good insight
Thank you

closed 2021-01-07 by smitkothari