linux-nvme/nvme-cli #853: Support for seperately compiled plugins

avylove, 2021-01-07:
As I understand it, plugins are compiled along with nvme-cli. Would it possible to support compiling the plugins separately and discovering them at runtime? This, of course, assumes they have been compiled against the same version of nvme-cli, similar to the way Linux kernel modules work. This would allow vendors to distribute binary plugins, which isn't ideal, but the reality is some vendors aren't willing to open source their code.

keithbusch, 2021-01-08:
Correct that "plugins" are all compiled in (so they probably should have been called vendor "extensions" rather than a plugin, but oh well).

It should be possible to extend this so we don't need to compile them in. We just need to define the API and dlopen shared objects installed in a well defined path. I can add it to my to-do list, but I can't say I'd be able to get around to it in the short term, but maybe next month.

igaw, 2022-01-19:
I am closing this issue now due to lacking progress. If anyone is willing to work on, please reopen or even better provide PR.

closed 2022-01-19 by igaw