linux-nvme/nvme-cli #856: Script to Check a Patch Satisfying min Coding Guidelines or Not

ganaidu707, 2021-01-08:
Could you tell us like what are the guidelines does nvme-cli follows? 

We can have a script to check whether the given patch following certain guidelines.

> e.g: 1. 80 Character per line
> e.g: 2. If any white space characters check etc..

or it is already existed?

keithbusch, 2021-01-08:
We've adopted the Linux kernel coding style, though they've been very loosely applied to the vendor extensions... I think the kernel's checkpatch.pl utility should work here.

ganaidu707, 2021-01-11:
Yes that works here, but not able to find that checkpatch.pl utility here, can we pull it to this code base, so that before commit, we can verify coding guidelines.

closed 2021-04-05 by ganaidu707