linux-nvme/nvme-cli #859: unable to read USB

JustSumDad, 2021-01-11:
I tested this under Rasperry pi OS on an RPI4
Using an NVME USB enclosure I was not able to detect the device.

Thanx

keithbusch, 2021-01-11:
Does the kernel even support this?

JustSumDad, 2021-01-20:
"Does the kernel even support this?"

I have no idea, I was hopeing someone else would chime in...
how would one look up if the kernel had this support?

keithbusch, 2021-01-20:
Knowing nothing about the device you are using, I'm going to take a wild guess that this "enclosure" doesn't appear to the host as an NVMe device since there's no "nvme-over-usb" protocol that I'm aware of, and instead appears as a mass storage device that subscribes to the kernel's scsi layer. If that's the case, then it will not apply to this nvme utility, but it should show up as a disk with a name like /dev/sdb (or sdc, sdd, etc...), and you should be able to see it from the output of 'lsusb' and 'lsscsi'.

JustSumDad, 2021-01-21:
Thanx much for the input!
my goal is just to get a temp reading of the drive like I can do via a windows system and CrystalDiskInfo.
What you have said makes sense thank you, I will check out those two suggestions.

keithbusch, 2021-01-21:
If I am correct in how it interfaces with the kernel, then you should be able to get this kind of data via 'smartctl'.

fixthething, 2021-03-02:
smartctl can talk to ATA drives through a USB bridge chip; the exact vendor-specific passthrough USB commands needed can be different for each bridge chip manufacturer/device. I suspect similar direct NVMe command passthrough may exist, but would be particular to the chip in whatever USB enclosure you're using. I think it would be cool to build in that sort of thing, but it would require some vendor-specific bridge chip knowledge.

hreinecke, 2022-01-16:
We will, once there is an NVMe-over-USB transport specification. Until then: sorry, no. nvme-cli is implementing the NVMe standard, and if there is no standard we can't implement it.

closed 2022-01-16 by hreinecke

RJVB, 2023-09-19:
As an unasked-for courtesy, here's some output from `smartctl` ;)

```
> sudo smartctl -d test /dev/sdc
smartctl 7.4 2023-08-01 r5530 [x86_64-linux-4.14.23-ck1-mainline-core2-rjvb] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

/dev/sdc [USB NVMe Realtek]: Device of type 'sntrealtek' [NVMe] detected
/dev/sdc [USB NVMe Realtek]: Device of type 'sntrealtek' [NVMe] opened

> smartctl 7.4 2023-08-01 r5530 [x86_64-linux-4.14.23-ck1-mainline-core2-rjvb] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       WD Blue SN570 1TB
Serial Number:                      23142G805228
Firmware Version:                   234110WD
PCI Vendor/Subsystem ID:            0x15b7
IEEE OUI Identifier:                0x001b44
Total NVM Capacity:                 1,000,204,886,016 [1.00 TB]
Unallocated NVM Capacity:           0
Controller ID:                      0
NVMe Version:                       1.4
Number of Namespaces:               1
Namespace 1 Size/Capacity:          1,000,204,886,016 [1.00 TB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            001b44 8b4a4c4c12
Local Time is:                      Wed Sep 20 01:05:01 2023 CEST
Firmware Updates (0x14):            2 Slots, no Reset required
Optional Admin Commands (0x0017):   Security Format Frmw_DL Self_Test
Optional NVM Commands (0x005f):     Comp Wr_Unc DS_Mngmt Wr_Zero Sav/Sel_Feat Timestmp
Log Page Attributes (0x1e):         Cmd_Eff_Lg Ext_Get_Lg Telmtry_Lg Pers_Ev_Lg
Maximum Data Transfer Size:         128 Pages
Warning  Comp. Temp. Threshold:     80 Celsius
Critical Comp. Temp. Threshold:     85 Celsius
Namespace 1 Features (0x02):        NA_Fields

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +     4.20W    3.70W       -    0  0  0  0        0       0
 1 +     2.70W    2.30W       -    0  0  0  0        0       0
 2 +     1.90W    1.80W       -    0  0  0  0        0       0
 3 -   0.0250W       -        -    3  3  3  3     3900   11000
 4 -   0.0050W       -        -    4  4  4  4     5000   44000

Supported LBA Sizes (NSID 0x1)
Id Fmt  Data  Metadt  Rel_Perf
 0 +     512       0         2
 1 -    4096       0         1

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02)
Critical Warning:                   0x00
Temperature:                        29 Celsius
Available Spare:                    100%
Available Spare Threshold:          10%
Percentage Used:                    0%
Data Units Read:                    96,185 [49.2 GB]
Data Units Written:                 2,028,951 [1.03 TB]
Host Read Commands:                 3,060,436
Host Write Commands:                63,437,826
Controller Busy Time:               50
Power Cycles:                       29
Power On Hours:                     4
Unsafe Shutdowns:                   10
Media and Data Integrity Errors:    0
Error Information Log Entries:      1
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0

Warning: NVMe Get Log truncated to 0x200 bytes, 0x200 bytes zero filled
Error Information (NVMe Log 0x01, 16 of 256 entries)
No Errors Logged

Warning: NVMe Get Log truncated to 0x200 bytes, 0x034 bytes zero filled
Self-test Log (NVMe Log 0x06)
Self-test status: No self-test in progress
No Self-tests Logged
```

I would assume that smartmontools use an internal library to present a single interface for sending smart requests to the device; would it help if they bundled that in a shared library?
Given how there is still no official SATA-over-USB standard (judging from the number of USB types smartmontools supports) it seems unlikely there will be something like that for NVME - unless the community drafts one.