linux-nvme/nvme-cli #862: Incorrect return code for IO command during drive in locked state.

lavin1993-spec, 2021-01-12:
**Issue:**
* Incorrect return code is seen for read or write command when issued in a locked drive(locked using opal commands).
* Expected (134) but returned (5).
* This unexpected return code is not seen in older nvme-cli version(1.6)
* Able to see this in latest versions(1.9, 1.13)

**NVMeCLI version 1.6**
sudo nvme --version;sudo nvme write-zeroes /dev/nvme0n1;echo $?
nvme version 1.6
NVME IO command error:ACCESS_DENIED: Access to the namespace and/or LBA range is denied due to lack of access rights(6286)
134

**NVMeCLI version 1.9**
sudo nvme --version;sudo nvme write-zeroes /dev/nvme0n1;echo $?
nvme version 1.9
NVMe status: ACCESS_DENIED: Access to the namespace and/or LBA range is denied due to lack of access rights(0x6286)
5

Is there any other standard way to get the return code across all version or 
will the upcoming version fix the above difference?

Thanks

keithbusch, 2021-01-12:
134 wasn't correct to begin with. The program used to exit with a nvme status sometimes, and other times with errno. The problem is exit status is truncated to 8 bits, but nvme status is 15 bits, which you can see is truncated nvme status 0x286, and we couldn't differentiate those from errnos, so now they're all errnos.

hreinecke, 2022-01-16:
So this is intentional, then. Closing.

closed 2022-01-16 by hreinecke