linux-nvme/nvme-cli #867: nvme discover doesn't support setting --keep-alive-tmo

yizhanglinux, 2021-01-14:
After commit[1], the discover operation will have one default kato value instead of zero,  and it will lead the discover cmd failed to one old target, from the commit description, seems we can pass a zero kato value to fix it. Then I tried [3], it still failed and seems nvme-cli doesn't support pass zero kato value, but from the help page[4] it shows it already support it, could you help check this problem? 

[1]
```
commit 2d352df57bcd7fee800fa1aa7257e68bfca64d2b
Author: Sagi Grimberg <sagi@grimberg.me>
Date:   Fri Jul 12 11:02:09 2019 -0700

    nvme-fabrics: allow discovery subsystems accept a kato
    
    This modifies the behavior of discovery subsystems to accept
    a kato as a preparation to support discovery log change
    events. This also means that now every discovery controller
    will have a default kato value, and for non-persistent connections
    the host needs to pass in a zero kato value (keep_alive_tmo=0).

```
[2]
```
$ nvme discover -t rdma -a 172.31.0.180 -c 1
Failed to write to /dev/nvme-fabrics: Input/output error
$ dmesg
[84229.797735] nvme nvme34: Connect command failed, error wo/DNR bit: 2
[84229.806448] nvme nvme34: failed to connect queue: 0 ret=16386

```
[3]
```
$ nvme discover -t rdma -a 172.31.0.180 -k 0
Failed to write to /dev/nvme-fabrics: Input/output error
```
[4]
```
$ ./nvme discover -h
Usage: nvme discover <device> [OPTIONS]

Send Get Log Page request to Discovery Controller.

Options:
  [  --transport=<LIST>, -t <LIST> ]    --- transport type
  [  --traddr=<LIST>, -a <LIST> ]       --- transport address
  [  --trsvcid=<LIST>, -s <LIST> ]      --- transport service id (e.g. IP
                                            port)
  [  --host-traddr=<LIST>, -w <LIST> ]  --- host traddr (e.g. FC WWN's)
  [  --hostnqn=<LIST>, -q <LIST> ]      --- user-defined hostnqn (if default
                                            not used)
  [  --hostid=<LIST>, -I <LIST> ]       --- user-defined hostid (if default
                                            not used)
  [  --raw=<LIST>, -r <LIST> ]          --- raw output file
  [  --device=<LIST>, -d <LIST> ]       --- use existing discovery controller
                                            device
  [  --keep-alive-tmo=<NUM>, -k <NUM> ] --- keep alive timeout period in
                                            seconds
  [  --reconnect-delay=<NUM>, -c <NUM> ] --- reconnect timeout period in
                                            seconds
  [  --ctrl-loss-tmo=<NUM>, -l <NUM> ]  --- controller loss timeout period in
                                            seconds
  [  --tos=<NUM>, -T <NUM> ]            --- type of service
  [  --hdr_digest, -g ]                 --- enable transport protocol header
                                            digest (TCP transport)
  [  --data_digest, -G ]                --- enable transport protocol data
                                            digest (TCP transport)
  [  --nr-io-queues=<NUM>, -i <NUM> ]   --- number of io queues to use
                                            (default is core count)
  [  --nr-write-queues=<NUM>, -W <NUM> ] --- number of write queues to use
                                            (default 0)
  [  --nr-poll-queues=<NUM>, -P <NUM> ] --- number of poll queues to use
                                            (default 0)
  [  --queue-size=<NUM>, -Q <NUM> ]     --- number of io queue elements to
                                            use (default 128)
  [  --persistent, -p ]                 --- persistent discovery connection
  [  --quiet, -S ]                      --- suppress already connected errors
  [  --matching, -m ]                   --- connect only records matching the
                                            traddr

```

keithbusch, 2021-01-14:
The error is coming from the kernel. What kernel version are you using?

yizhanglinux, 2021-01-17:
Hi Keith
The target is using 4.20,  I did some testing, finally found bellow patch can fix the issue on target side, after this patch, the target side can accept non-zero kato which pass from host. 
I'm thinking if it's reasonable if we can enable nvme-cli to set kato:0, so it will works with old target. and from sagi's comment [2], seems we can pass zero kato to target.

[1]
```
commit f9362ac1738a41cc526fde82e76beb034d8c6053
Author: Jay Sternberg <jay.e.sternberg@intel.com>
Date:   Mon Nov 12 13:56:35 2018 -0800

    nvmet: allow Keep Alive for Discovery controller
    
    Per change to specification allowing Discovery controllers to have
    explicit persistent connections, remove restriction on Discovery
    controllers allowing kato on connect.
    
    Signed-off-by: Jay Sternberg <jay.e.sternberg@intel.com>
    Reviewed-by: Hannes Reinecke <hare@suse.com>
    Reviewed-by: Sagi Grimberg <sagi@grimberg.me>
    Reviewed-by: Christoph Hellwig <hch@lst.de>
    Signed-off-by: Christoph Hellwig <hch@lst.de>
    Signed-off-by: Jens Axboe <axboe@kernel.dk>

diff --git a/drivers/nvme/target/admin-cmd.c b/drivers/nvme/target/admin-cmd.c
index 2e89f4e3364b..ad94119f6a51 100644
--- a/drivers/nvme/target/admin-cmd.c
+++ b/drivers/nvme/target/admin-cmd.c
@@ -726,7 +726,7 @@ static void nvmet_execute_async_event(struct nvmet_req *req)
        schedule_work(&ctrl->async_event_work);
 }
 
-static void nvmet_execute_keep_alive(struct nvmet_req *req)
+void nvmet_execute_keep_alive(struct nvmet_req *req)
 {
        struct nvmet_ctrl *ctrl = req->sq->ctrl;
 
diff --git a/drivers/nvme/target/core.c b/drivers/nvme/target/core.c
index f42a105ef17f..46345c3f9d4a 100644
--- a/drivers/nvme/target/core.c
+++ b/drivers/nvme/target/core.c
@@ -1180,31 +1180,17 @@ u16 nvmet_alloc_ctrl(const char *subsysnqn, const char *hostnqn,
        ctrl->cntlid = ret;
 
        ctrl->ops = req->ops;
-       if (ctrl->subsys->type == NVME_NQN_DISC) {
-               /* Don't accept keep-alive timeout for discovery controllers */
-               if (kato) {
-                       status = NVME_SC_INVALID_FIELD | NVME_SC_DNR;
-                       goto out_remove_ida;
-               }
 
-               /*
-                * Discovery controllers use some arbitrary high value in order
-                * to cleanup stale discovery sessions
-                *
-                * From the latest base diff RC:
-                * "The Keep Alive command is not supported by
-                * Discovery controllers. A transport may specify a
-                * fixed Discovery controller activity timeout value
-                * (e.g., 2 minutes).  If no commands are received
-                * by a Discovery controller within that time
-                * period, the controller may perform the
-                * actions for Keep Alive Timer expiration".
-                */
-               ctrl->kato = NVMET_DISC_KATO;
-       } else {
-               /* keep-alive timeout in seconds */
-               ctrl->kato = DIV_ROUND_UP(kato, 1000);
-       }
+       /*
+        * Discovery controllers may use some arbitrary high value
+        * in order to cleanup stale discovery sessions
+        */
+       if ((ctrl->subsys->type == NVME_NQN_DISC) && !kato)
+               kato = NVMET_DISC_KATO_MS;
+
+       /* keep-alive timeout in seconds */
+       ctrl->kato = DIV_ROUND_UP(kato, 1000);
+
        nvmet_start_keep_alive_timer(ctrl);
 
        mutex_lock(&subsys->lock);
@@ -1215,8 +1201,6 @@ u16 nvmet_alloc_ctrl(const char *subsysnqn, const char *hostnqn,
        *ctrlp = ctrl;
        return 0;
 
-out_remove_ida:
-       ida_simple_remove(&cntlid_ida, ctrl->cntlid);
 out_free_sqs:
        kfree(ctrl->sqs);
 out_free_cqs:
diff --git a/drivers/nvme/target/discovery.c b/drivers/nvme/target/discovery.c
index bc0aa0bf1543..6e0f91eb34ce 100644
--- a/drivers/nvme/target/discovery.c
+++ b/drivers/nvme/target/discovery.c
@@ -194,6 +194,10 @@ u16 nvmet_parse_discovery_cmd(struct nvmet_req *req)
        }
 
        switch (cmd->common.opcode) {
+       case nvme_admin_keep_alive:
+               req->execute = nvmet_execute_keep_alive;
+               req->data_len = 0;
+               return 0;
        case nvme_admin_get_log_page:
                req->data_len = nvmet_get_log_page_len(cmd);
 
diff --git a/drivers/nvme/target/nvmet.h b/drivers/nvme/target/nvmet.h
index 8ddc54fa98c7..c525fb0510c5 100644
--- a/drivers/nvme/target/nvmet.h
+++ b/drivers/nvme/target/nvmet.h
@@ -372,6 +372,8 @@ void nvmet_req_complete(struct nvmet_req *req, u16 status);
 int nvmet_req_alloc_sgl(struct nvmet_req *req);
 void nvmet_req_free_sgl(struct nvmet_req *req);
 
+void nvmet_execute_keep_alive(struct nvmet_req *req);
+
 void nvmet_cq_setup(struct nvmet_ctrl *ctrl, struct nvmet_cq *cq, u16 qid,
                u16 size);
 void nvmet_sq_setup(struct nvmet_ctrl *ctrl, struct nvmet_sq *sq, u16 qid,
@@ -442,7 +444,7 @@ u32 nvmet_get_log_page_len(struct nvme_command *cmd);
 #define NVMET_DEFAULT_ANA_GRPID        1
 
 #define NVMET_KAS              10
-#define NVMET_DISC_KATO                120
+#define NVMET_DISC_KATO_MS             120000
 
 int __init nvmet_init_configfs(void);
 void __exit nvmet_exit_configfs(void);
```

[2]
```
commit 2d352df57bcd7fee800fa1aa7257e68bfca64d2b
Author: Sagi Grimberg <sagi@grimberg.me>
Date:   Fri Jul 12 11:02:09 2019 -0700

    nvme-fabrics: allow discovery subsystems accept a kato
    
    This modifies the behavior of discovery subsystems to accept
    a kato as a preparation to support discovery log change
    events. This also means that now every discovery controller
    will have a default kato value, and for non-persistent connections
    the host needs to pass in a zero kato value (keep_alive_tmo=0).
```

Thanks
Yi

keithbusch, 2021-01-19:
Yeah, that's a really old kernel for your target. I don't think the capability was added until something like 5.4 where the patch you located initially landed.

closed 2021-01-19 by keithbusch