linux-nvme/nvme-cli #878: set feature doesn't work with APST

xanus-7, 2021-01-26:
Hi, I 'm trying to disable APST with set feature command, but it doesn't work.

My ssd:

 /dev/nvme0n1 17A807981DBE00292731 SPCC M.2 PCIe SSD 

The command I'm running:

sudo nvme set-feature -f 0x0c -v=0 /dev/nvme0n1

I get this result:

set-feature:0c (Autonomous Power State Transition), value:00000000

But if I check APST with sudo nvme get-feature -f 0x0c -H /dev/nvme0n1

get-feature:0xc (Autonomous Power State Transition), Current value:0x000001
Autonomous Power State Transition Enable (APSTE): Enabled

So it's still enabled. Any ideas?

keithbusch, 2021-01-26:
There are some odd things in your output, but as far as disabling apst, doing this from user space is not the correct way to achieve that: the kernel's default policy is to enable APST, so whatever you do from user space is going to be undone by the kernel when it rechecks your controller.

To properly disable APST such that the kernel will not be re-enabling it, you need to use the kernel parameter `nvme_core.default_ps_max_latency_us=0`.

Now it does seem suspicious that your device is returning the wrong value immediately; the kernel's own settings don't typically take effect without some event triggering it to rescan your device. So the fact getting the feature returns the unexpected value so soon appears suspect on the device side.

xanus-7, 2021-01-26:
I tried nvme_core.default_ps_max_latency_us=0 already before meddling with nvme-cli, it doesn't disable APST.  Applied it at boot.
So what's going on here?

keithbusch, 2021-01-27:
This is starting to sound like a device and/or platform issue. With your kernel parameter applied, can you confirm that `cat /sys/module/nvme_core/parameters/default_ps_max_latency_us` prints `0`?

xanus-7, 2021-01-27:
Prints 0, I'm still getting this bug though:

https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1678184

Which is the reason, I wanted to disable APST in the first place.

keithbusch, 2021-01-27:
You might want to file a bug with your nvme device vendor.

xanus-7, 2021-01-28:
Not unless I learn Chinese.

closed 2021-01-28 by xanus-7

keithbusch, 2021-01-28:
If that's really true, it sounds like your vendor's internationalization efforts are unworthy of international consumers.

cipianpascu, 2021-05-29:
I have ubuntu 20.04 installed on a Huawei matebook x pro. I struggled with the original Toshiba nvme and I replaced it with Samsung 970 pro. I get the same readonly issue and I'm unable to stop the APST feature.

mangomangomango0820, 2022-10-29:
> Not unless I learn Chinese.

hey bro .. 
I was searching for some info due to a issue and happend to find this issue (that's why I am leaving a comment in 2022 to an issue put up in 2021.). 

I am wondering which nvme device you're using ...
(If you do not mind, I suggest using lspci | grep Non to capture the BDF of you ssd and then lspci -n | grep BDF to get the vendor id ...)

IvnSoft, 2023-07-28:
Sorry to comment on an old issue.
Im getting the same issue. When using nvme set-feature the command hangs. No errors logged. Ctrl+C works.

Ive tried everything to disable APST , and various values for nvme_core.default_ps_max_latency_us (5500,200,0, and some random ones), and nothing.

Now, im not even sure i need to disable APST, but im getting system freezes with these drives (Samsung 970 and 970+, and a Micron 512GB). They are fine, and work perfectly in any other system (or with an NGFF-USB adapter). RAM is sane, and everything else checks out. The laptop is an Asus TUF A15 FA506IV (320 firmware)

The ONLY thing different, is these drives with the issue, are NVME 1.3, whereas the main one, is NVME 1.4.
Every single time i get a freeze, a counter for an invalid NVME command (status 0x4004 or 0x4005) goes up by 1.

04:00.0 Non-Volatile memory controller: Sandisk Corp WD Blue SN550 NVMe SSD (rev 01)
05:00.0 Non-Volatile memory controller: Micron Technology Inc Device 5404 (rev 03)